magic-blue-smoke/Dual-Edge-TPU-Adapter #2: Testers are needed for Dual Edge TPU Adapter board prototypes

magic-blue-smoke, 2021-06-29:
Testers are needed for Dual Edge TPU Adapter board prototypes.

Boards available:
* m.2 2242* AE key
* m.2 2280* BM key (can be dremeled down to 2242, mounting holes at 42, 60 and 80)
(*) actual width is 24mm

DM on twitter (preferred) @magic__smoke (double underscore) if interested or leave a message here.

mattncsu, 2021-06-30:
Is there a M.2. to PCIe or USB3 adapter that would work with it? I have an empty M.2 M key slot in my laptop meant for nvme storage but not sure if it can take non-storage peripherals.

pdecat, 2021-06-30:
Hi @magic-blue-smoke, I'm interested and currently looking for the best option to get my Dual Edge Coral fully working in a PCIe slot ([only have one core visible right now](https://github.com/google-coral/edgetpu/issues/256#issuecomment-747942260)).

Do you believe your m.2 E key to m.2 BM key adapter would work with this m.2 M Key to x4 PCIe? https://www.amazon.com/Cablecc-Carte-Adaptateur-M-Key-Express/dp/B00S6B6MZW/

magic-blue-smoke, 2021-06-30:
@mattncsu 
> Is there a M.2. to PCIe or USB3 adapter that would work with it? I have an empty M.2 M key slot in my laptop meant for nvme storage but not sure if it can take non-storage peripherals.
These adapters are likely not to fit laptops, since adapter width is 24mm, while standard is 22mm. Another issue is that m.2 connector can be SATA instead of nvme/PCIe. I didn't have laptops in mind while making prototypes, thinking target systems are SBCs. I just ordered random m.2 M Key to x4 PCIe adapter, hopefully can test tomorrow.

@pdecat 
> Do you believe your m.2 E key to m.2 BM key adapter would work with this m.2 M Key to x4 PCIe? https://www.amazon.com/Cablecc-Carte-Adaptateur-M-Key-Express/dp/B00S6B6MZW/
Hi, just ordered random m.2 M Key to x4 PCIe adapter to test tomorrow

saket424, 2021-06-30:
@magic-blue-smoke 
I would be interested in the m.2 2280 BM key version of the adapter to be mated with the odyssey blue available from seedstudio or amazon

https://www.seeedstudio.com/Odyssey-Blue-J4125-128GB-p-4921.html
https://www.amazon.com/Mini-Odyssey-Blue-J4105-Bluetooth5-0/dp/B08FD3QHQC

Let me know how I can get one to test and evaluate. thanks

mattncsu, 2021-06-30:
I'm using an Atomic Pi SBC with a USB3 coral now but I have a few idle desktops with open PCIe slots that I could migrate my code to.  Also hoping your board would allow a USB3 adapter such as one of these ([link1](https://www.amazon.com/NGFF-Adapter-6pin-Slot-Module/dp/B07PDJPP4H) or [link2](https://www.amazon.com/Adapter-RIITOP-Reader-Compatible-Based/dp/B08FWZXT2N)) to be used with

magic-blue-smoke, 2021-06-30:
@saket424 
> @magic-blue-smoke
> I would be interested in the m.2 2280 BM key version of the adapter to be mated with the odyssey blue available from seedstudio or amazon [...]

I'll collect requests till Sunday to know how many boards to assemble, then collect addresses and ship

@mattncsu 
> I'm using an Atomic Pi SBC with a USB3 coral now but I have a few idle desktops with open PCIe slots that I could migrate my code to. Also hoping your board would allow a USB3 adapter such as one of these ([link1](https://www.amazon.com/NGFF-Adapter-6pin-Slot-Module/dp/B07PDJPP4H) or [link2](https://www.amazon.com/Adapter-RIITOP-Reader-Compatible-Based/dp/B08FWZXT2N)) to be used with the Atomic Pi or RPi4

m.2 connector depending on its type carries variety of interfaces:
Link1 - provides USB directly to m.2 connector for LTE modems
Link2 - USB to SATA connector
None of these provide PCIe and will not work with my adapters

Regarding Dual Edge to PCIe (desktop) - this can be easily done later this year and I will make those if AE and BM versions would meet some demand. Going to test if my BM key adapter works on desktop with nvme-to-PCIe board tomorrow.

magic-blue-smoke, 2021-07-02:
@pdecat @mattncsu @saket424 
[Tested BM adapter on PCIe to m.2 M key adapter](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/1#issuecomment-872631477)
[Updated main page with BM photo](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter)

mattncsu, 2021-07-02:
Glad it works with the PCIe adapter.  Wonder if one of these would work-https://www.amazon.com/Cablecc-Carte-Adaptateur-M-Key-Express/dp/B00S6B7T22/ could squeze 8 TPUs on one card

magic-blue-smoke, 2021-07-02:
@mattncsu 
> Glad it works with the PCIe adapter. Wonder if one of these would work-https://www.amazon.com/Cablecc-Carte-Adaptateur-M-Key-Express/dp/B00S6B7T22/ could squeze 8 TPUs on one card

It should work with some exceptions:
- Motherboard must support PCIe x16 bifurcation to 4x PCIe x4 (as on 3rd picture in description)
- Clearance between m.2 connectors is unclear from those pictures. My adapter has 24mm width instead of 22 to ease layout and multiple adapters might not fit. I can fix it in a next revision

mattncsu, 2021-07-02:
Looks like my Z170 based board is too old to support bifurcation. Just being able to run a dual tpu in one slot will likely be enough for my needs anyway.

magic-blue-smoke, 2021-07-08:
Update: 
I soldered 5 BM adapter boards. It was something I'd like to be done by pick-and-place machine next time. 
Going to test adapters tomorrow.

magic-blue-smoke, 2021-07-10:
Please let me know your shipping address via twitter DM or leave contact email here

mattncsu, 2021-07-10:
@magic-blue-smoke matt-git@teammakeithappen.com

TheOfficialMrBlah, 2021-07-12:
Hi,
I am interested in such an adapter (m.2 m-Key).
Is it still possible to test one?
I would like to install a Dual Edge TPU in my PC or server and be able to use both cores.

One question, purely out of interest:
Is it possible to use this adapter in an M.2 to USB enclosure for nvme SSDs?

magic-blue-smoke, 2021-07-12:
@TheOfficialMrBlah 
I had 5 PCBs, all reserved, but one is missing shipping address. If it will not be confirmed in a week, I’ll consider shipping it to another reviewer 

External m.2 USB enclosure: 99.99% this is SATA, not NVMe (PCIe) enclosure. 
m.2 M key has four configuration pins to tell host system what interfaces should be used, PCIe and SATA are among them, but SATA enclosure simply has no PCIe.
Is there a chance to use it with external enclosure? Maybe. If some laptops allow PCIe over USB-C (or over Thunderbolt?) then there are chances for adapter ending with PCIe connector

magic-blue-smoke, 2021-07-12:
@TheOfficialMrBlah 
Update: I expect EU reviews are getting adapters this week. If results are promising, I’ll make desktop PCIe x1 to Coral Dual TPU adapter, this is where another samples will be available

TheOfficialMrBlah, 2021-07-12:
Thanks for the answer!

Both adapters sound interesting, so if there's something to test again, I'm there. 
___

I'm not so sure about the USB adapter.
I looked up the chip, it's a JMicron JMS583, which provides PCIe Gen3 x2 according to the data sheet.

The Dual Edge TPU with your adapter in such a housing would also be interesting, it would be a good and more powerful alternative to the Coral USB Accelerator.

magic-blue-smoke, 2021-07-12:
@TheOfficialMrBlah I have a mixed feelings regarding this adapter. One one hand it has PCIe, on another it has “Mass Storage” in description and that gives me idea that it only translates USB mass storage class device commands and transfers to NVMe commands. Maybe someone who will receive adapter samples can test this (as I’m sending all adapters I had) but my expectations regarding this particular enclosure  are quite low.

TheOfficialMrBlah, 2021-07-12:
> @TheOfficialMrBlah I have a mixed feelings regarding this adapter. One one hand it has PCIe, on another it has “Mass Storage” in description and that gives me idea that it only translates USB mass [...]

I also have doubts that it will work, because the adapter, as you wrote, is designed for mass storage.
But it would be interesting if someone could test it.

magic-blue-smoke, 2021-07-22:
Please share your test results and findings here: /issues/3

vishalchaudhary, 2021-07-23:
@magic-blue-smoke I would buy the first adapter for M.2 M Key to the M.2 E Key for the google coral dual edge tpu. Let me know if you still need testers/ early adopters. I am from India and ready to even pay for the shipping of the prototype.

You are doing an amazing job for enthusiasts like us.

magic-blue-smoke, 2021-07-23:
Hi @vishalchaudhary 
I mailed all B+M key adapters I had. With successful tests I’ll proceed with production batch. 
Thank you for supportive words and consider following project here and on Twitter

digiblur, 2021-08-12:
When you get another batch of things, I'd be interested in purchasing one.  Glad to see some support for this!

wmchris, 2021-08-24:
Small update: https://de.aliexpress.com/item/1005002596901609.html with bifurcation is working as intended. Just make sure your mainboard has bifurcation support

closed 2021-09-03 by magic-blue-smoke

sojab0on, 2025-08-10:
Happy to test one in a Synology rs unit if still needed