magic-blue-smoke/Dual-Edge-TPU-Adapter #4: PCIe x1 version pre-order

magic-blue-smoke, 2021-08-05:
PCIe x1 (desktop) version is planned to be available for purchase around September.
To estimate production volume, please DM me on Twitter @magic__smoke (double underscore) with boards quantity.

TheStealth359, 2021-08-10:
Hi,

I would be interested on one board depending on cost. (sorry don't have a Twitter account)

magic-blue-smoke, 2021-08-10:
> Hi,
> 
> I would be interested on one board depending on cost. (sorry don't have a Twitter account)

@TheStealth359 For low-profile PCIe x1 desktop adapter I‘m targeting at $30 shipped and will post online store link here

I’m also looking if someone would be interested in PCIe x4 carrier for 8 TPUs (four Dual Edge TPU cards)

TheStealth359, 2021-08-10:
> 
> @TheStealth359 For low-profile PCIe x1 desktop adapter I‘m targeting at $30 shipped and will post online store link here
> 
> I’m also looking if someone would be interested in PCIe x4 carrier for 8 TPUs (four Dual Edge TPU cards)

@magic-blue-smoke Thanks for the info, I'll keep my eyes open for the link. As for a board for 8 TPUs, I don't have a use case for that at the moment.

Danny2100, 2021-08-13:
> @TheStealth359 For low-profile PCIe x1 desktop adapter I‘m targeting at $30 shipped and will post online store link here
>
> I’m also looking if someone would be interested in PCIe x4 carrier for 8 [...]

Just to be sure as I'm not familiar with the terminology. The pcie x1 version is compatible with the coral m.2 dual edge tpu?

Do you have an idea of when the link will be available to purchase?

Not that I'm sure I have a use case for the pcie x4 that can handle 4 m.2 dual edge tpus but how much would that cost?

magic-blue-smoke, 2021-08-13:
> Just to be sure as I'm not familiar with the terminology. The pcie x1 version is compatible with the coral m.2 dual edge tpu?

@Danny2100 
You're right, I should make kinda FAQ at the front page
In this repo we're talking about using Coral Dual Edge TPU on a systems which don't have m.2 E-key (or E-key slot is not to full specification):
- PCIe x1 adapter is to connect Coral Dual Edge TPU to desktop PCIe x1 slot
- m.2 BM adapter is to connect Coral Dual Edge TPU to a system with m.2 B or M key slot. **Important** it has to be m.2 slot for NVMe (PCIe) drives. Slots for SATA drives and USB-SATA enclosures will not work. Also, I **don't** expect USB to m.2 B/M NVMe enclosures to work as well
- m.2 AE adapter is to connect Coral Dual Edge TPU to a system with m.2 A or E key. If you have E-key make sure you can only see one core, otherwise you're lucky and don't need adapter at all

In all of cases above, there must be PCIe on a slot. For example, m.2 slot for LTE modem might only have USB and no PCIe

> Do you have an idea of when the link will be available to purchase?

Mid- to end of September for desktop low profile PCIe x1 adapter
m.2 AE and BM key adapters are on hold, as almost all BM key test units were used in PCs and desktop PCIe slots. 

> Not that I'm sure I have a use case for the pcie x4 that can handle 4 m.2 dual edge tpus but how much would that cost?

Several hundreds, but well below than solution from Asus.

Danny2100, 2021-08-13:
> @Danny2100
> You're right, I should make kinda FAQ at the front page
> In this repo we're talking about using Coral Dual Edge TPU on a systems which don't have m.2 E-key (or E-key slot is not to full [...]

Perfect I'm definitely interested in the pcie x1 and I just ordered a coral dual edge tpu in advance. Are your products shipping form somewhere in the US?

magic-blue-smoke, 2021-08-13:
> Perfect I'm definitely interested in the pcie x1 and I just ordered a coral dual edge tpu in advance. Are your products shipping form somewhere in the US or internationally?

$30 includes standard international shipping with tracking number. Transit time to address in the US is 8-14 days.

wmchris, 2021-08-14:
> $30 includes standard international shipping with tracking number. Transit time to address in the US is 8-14 days.

thats a very fair price - especially if you consider the high costs for worldwide shipping.

Danny2100, 2021-08-14:
> > $30 includes standard international shipping with tracking number. Transit time to address in the US is 8-14 days.
> 
> thats a very fair price - especially if you consider the high costs for worldwide shipping.

Yep I'm in the US I just wasn't sure where it was shipping from.

wmchris, 2021-08-14:
> Yep I'm in the US I just wasn't sure where it was shipping from.
i got mine from the european union, so chances are high that the retail ones will be shipped from there, too.

magic-blue-smoke, 2021-08-14:
> i got mine (the pre-release version) from the european union, so chances are high that the retail ones will be shipped from there, too.

Production units will be shipped from Russia.
I’m open to partner with local EU stores and that makes sense for buyers after EU eliminated €22 no VAT threshold on imported items

tikwanleap, 2021-08-16:
I'm interested in the pcie x1 also.

retrospectsecurity, 2021-08-18:
I am interested in one for embedded boards.

magic-blue-smoke, 2021-08-18:
> I'm interested in the pcie x1 also.

@tikwanleap thanks, good to know. Doing my best to start shipping in late September 

> I am interested in one for embedded boards.

Please let me know which key and size (or target main board) you’re interested in. 
m.2 versions are in extremely low demand and I was planning to make those available around December. Earlier availability is possible with paid samples or with MOQ

treichst, 2021-08-19:
I am very interested in a M > E Key adapter to get both TPUs to work

I already have the Dual TPU and https://www.amazon.com/dp/B00S6B6MZW?psc=1&ref=ppx_yo2_dt_b_product_details 
I have confirmed that my motherboard supports bifurcation.

magic-blue-smoke, 2021-08-19:
> I am very interested in a M > E Key adapter to get both TPUs to work
> I already have the Dual TPU and https://www.amazon.com/dp/B00S6B6MZW?psc=1&ref=ppx_yo2_dt_b_product_details
> I have confirmed that my motherboard supports bifurcation.

@treichst 
If you plan to use Dual Edge TPU in desktop PCIe x1 slot, I'd recommend you to use "Low profile PCIe x1 version" (first picture on front page/README.md). It will be available in September and doesn't need additional adapter you linked

If you still prefer m.2 BM adapter, it might be available around December and will be a bit more expensive than Low profile PCIe x1 adapter

PCIe bifurcation only makes sense if someone would like to try four Dual Edge TPUs with four m.2 BM adapters on this adapter card: https://www.amazon.com/Cablecc-Carte-Adaptateur-M-Key-Express/dp/B00S6B7T22/ to get eight TPUs. This config is theoretical and not yet tested

treichst, 2021-08-19:
Thank you for your reply - that helps clear things up a lot. 
The first picture appears to be exactly what I need.

treichst, 2021-08-19:
> @treichst
> If you plan to use Dual Edge TPU in desktop PCIe x1 slot, I'd recommend you to use "Low profile PCIe x1 version" (first picture on front page/README.md). It will be available in September [...]

Thank you for your reply - that helps clear things up a lot.
The first picture appears to be exactly what I need.

magic-blue-smoke, 2021-08-20:
> Thank you for your reply - that helps clear things up a lot.
> The first picture appears to be exactly what I need.

@treichst Thanks, I really need to make a small "50 shades of m.2" write up and compatibility table on Coral Dual Edge TPU

ozett, 2021-08-21:
I would be interested 1-2 boards for the dual TPU with e-key.
what do i have to do? listen to this thread? Waiting for a shoping site?

magic-blue-smoke, 2021-08-21:
> I would be interested 1-2 boards for the dual TPU with e-key.
> what do i have to do? listen to this thread? Waiting for a shoping site?

@ozett @retrospectsecurity I've opened /issues/5 to estimate quantity and ratio between different m.2 variants. Please let me know there which variant would work for you 

I plan to make those available around December and link will be posted there as well

bennyryan, 2021-08-22:
Very interested in getting one of these. Let me know when the pre-order is up.

ozett, 2021-08-22:
i dont want to spoil [issue5 ](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/5)with basic question:

i need an adapter for [Dual TPU](https://coral.ai/products/m2-accelerator-dual-edgetpu/#tech-specs) wie E-Key (not A-E, what is different, i guess) to connect Coral Dual Edge TPU to a usual [desktop PCIe slot](https://de.wikipedia.org/wiki/PCI_Express).

how would you call your adapter? or are there more adapter after each other needed? one single piece would be great...

wmchris, 2021-08-22:
> i dont want to spoil [issue5 ](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/5)with basic question:
> 
> i need an adapter for [Dual TPU](https://coral.ai/products/m2-accelerator-dual-edgetpu/#tech-specs) wie E-Key (not A-E, [what is different, i guess](https://de.wikipedia.org/wiki/M.2))
> to connect Coral Dual Edge TPU to a usual [desktop PCIe slot](https://de.wikipedia.org/wiki/PCI_Express).
> 
> how would you call your adapter? or are there more adapter after each other needed? one single piece would be great...
> 
> maybe helpful to assign your adapter to a TPU-Model/ or Key-type? or vice versa?

the a+e key is compatible with a and e keyed cards, so the accelerator will fit. 

the adapter is more than a rerouting of the lanes - it includes a pcie hub, so you can use the m.2 dual edge tpu, which requires 2 pcie lanes on a single pcie lane on your host system.

ozett, 2021-08-22:
i am not that familiar with m2 devices and key-options. but as far as i read on the internet, this :

> the a+e key is compatible with a and e keyed cards, so the accelerator will fit.

it is not the general case for available adapters that a+e slot takes also all e-cards without hassle. or am i wrong?
that let me thinking that one need **an explicit** e-slot "only" adapter for the dual TPU m.2 e-key adapter. but also wrong?
 
>if you buy the pcie version you should not need any additional adapters.

what is meant by that statement? i bought a mini-PCIe coral TPU which fittet in a mini-PCIe slot on my mainboard and worked.
i am searching for a solution to easyly fit coral-TPUs into PCI slot.
it seems to me, that a good general solution for multiple TPUs would be e-key adapter to PCI-slots to fit the Dual TPU.
so that i could have 4 TPUS with 2 adapters?

magic-blue-smoke, 2021-08-22:
To all: I've updated README.md and do hope it made things more clear

@ozett to use Coral Dual Edge TPU with desktop PCIe x1 slot you'd need my "Low profile PCIe x1" adapter. Adapter can carry only one Coral Dual Edge TPU card (with two TPUs). To have more TPUs, more adapter cards can be installed. 

I guess that by saying "if you buy the pcie version you should not need any additional adapters" @wmchris meant purchasing my "Low profile PCIe x1" adapter.

If you need more TPUs, I'd like to hear if someone is interested in PCIe x4 card to carry four Dual Edge TPUs (eight TPUs total)

ozett, 2021-08-23:
thanks for clarification.

> If you need more TPUs, I'd like to hear if someone is interested in PCIe x4 card to carry four Dual Edge TPUs (eight TPUs total)

me 😄

wmchris, 2021-08-23:
> if e-key connetor suits to ae-key slos, what is the use of [your adapater](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter#m2-ae-version) "Dual TPU with e-key to ae-key"

okay a little write up:
an A+E keyed slot will always be compatible with all A and E cards when implemented by the specifications.
on most mainboards the native A+E slot is NOT implemented by the specifications - the specification states, that it has 2 PCIe Lanes assigned to the port. As PCIe Lanes are rare on mainboards and in fact only very few cards really need them and it is generally marketed as "wireless card slot" the mainboard will only have one pcie lane there.

if you insert the dual edge card in this slot it will only offer you one of the two coral TPUs.

magics adapter will fix this by adding a pcie bridge on the adapter itself, so it will convert X1 to X2. obviously it has only X1 speed which has to be shared by both TPUs, but as far as i know in most cases this will rarely make any difference in performance.

magic-blue-smoke, 2021-08-23:
> thanks for clarification.
> 
> [but from previous post](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/4#issuecomment-903251620):
> if e-key connetor suits to ae-key slos, what is the use of [your adapater](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter#m2-ae-version) "Dual TPU with e-key to ae-key"

@wmchris is right on most things: if you can see only one TPU of your Dual Edge TPU, then E-key m.2 slot is not implemented to a full specs and adapter would enable second TPU. The only thing I'd like to correct is that Dual Edge TPU needs two independent PCIe x1 busses, not two lanes 

> > If you need more TPUs, I'd like to hear if someone is interested in PCIe x4 card to carry four Dual Edge TPUs (eight TPUs total)
> 
> me smile
> (minor question: is there a significant benefit for your Card over using Coral TPU with M-Key[ in multiple usual M-Key adapters](https://www.personal-view.com/talks/discussion/21377/putting-multiple-nvme-m2-ssds-/p1) ([ASUS](https://www.asus.com/Motherboards-Components/Motherboards/Accessories/HYPER-M-2-X16-CARD-V2/techspec/))- if that is even possible.. thinking )

[image]
I'd expect this card to work with four m.2 E-key Coral Edge TPU (4 TPUs) or four Coral Dual Edge TPU with four my m.2 BM key adapters (8 cores total, but looks like ugly adapter stack). I don't see dedicated power supply for 3.3 rail there, so it raises some questions and you must connect SATA power cable (SATA cable connected should be fine with non-Dual TPUs)

[image]
This card uses CPU feature known as PCIe bifurcation, ie PCIe x16 split into 4x PCIe x4. You can identify these cards by absence of PCIe switch and presence of PCIe clock fanout driver IC. Theoretically it might work with four m.2 E-key Coral Edge TPU (4 TPUs) or four Coral Dual Edge TPU with four my m.2 BM key adapters, but:
- CPU must support PCIe bifurcation
- this one from Asus clearly states that only NVMe cards can be used with this adapter. I don't know if it's true and PCIe bifurcation only works for NVMe drives or it's just their engineering department don't want to be flooded with support questions when people start to plug everything-with-m.2-e-key-connector in there 

> 1. How do i order a "Low profile PCIe x1" adapter to Germany? Price?

I'll post ordering info here and on Twitter. Price is expected to be $30 shipped. I'm not registered for IOSS, so there will be import VAT for EU

> 2. How do i order a "PCIe x4 card to carry four Dual Edge TPUs" to germany ? Price?

That's under development, I'll email you with detials

hagoston, 2021-08-23:
Will the adapter card be backward compatible with PCIe x1 version 2.0? Thanks

ozett, 2021-08-23:
> > 1. How do i order a "Low profile PCIe x1" adapter to Germany? Price?
> 
> I'll post ordering info here and on Twitter. Price is expected to be $30 shipped. I'm not registered for IOSS, so there will be import VAT for EU
> 
> > 1. How do i order a "PCIe x4 card to carry four Dual Edge TPUs" to germany ? Price?
> 
> That's under development, I'll email you with detials

that sound great! thanks a lot for all the information! 🎂

magic-blue-smoke, 2021-08-23:
> Will the adapter card be backward compatible with PCIe x1 version 2.0? Thanks

@hagoston Coral Dual Edge TPU and IC used in my adapter are PCIe Gen 2

Here are some test results demonstrating that single PCIe Gen2 lane is not bottleneck and inference time increase is not dramatic: /issues/3#issuecomment-885925888
Please not that Y-axis (inference time) starts with 6ms, not 0, otherwise this graph gives wrong impression at first sight

hagoston, 2021-08-23:
> > Will the adapter card be backward compatible with PCIe x1 version 2.0? Thanks
> 
> @hagoston Coral Dual Edge TPU and IC used in my adapter are PCIe Gen 2
> 
> Here are some test results demonstrating that single PCIe Gen2 lane is not bottleneck and inference time increase is not dramatic: [#3 (comment)](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/3#issuecomment-885925888)
> Please not that Y-axis (inference time) starts with 6ms, not 0, otherwise this graph gives wrong impression at first sight
> 
> Update: I was trying to guess where you can have Gen2 PCIe and just in case you’re going to use it with Raspberry Pi CM4/CMIO - it will not work

Great, thanks for the information.

I have an old [asus motherboard](https://www.cnet.com/products/asus-e35m1-m-pro-motherboard-micro-atx-amd-e-350-series/) and it has only gen 2.0 slots (PCIe 2.0 x16 and PCIe 2.0 x1). My plan is to run [Frigate NVR](https://github.com/blakeblackshear/frigate) on it.

ColeMundus, 2021-08-24:
I would like one.

truncs, 2021-08-31:
Interested in getting one as well!

jonathanschulberger, 2021-09-01:
Definitely interested in one PCIE x1 desktop adapter. I would also be interested in one x4 Dual Edge TPU depending on the estimated price!

jeramiapoland, 2021-09-01:
I'm interested in one PCIe x1 adapter and a m.2 AE adapter.

magic-blue-smoke, 2021-09-03:
Everything is in the hands of global logistics now. 
Preliminary, boards will be listed on Tindie and I'll separate board price from shipping price to make it fair to those who's ordering more than one board.

Danny2100, 2021-09-03:
When do you think they will be available for purchase?

magic-blue-smoke, 2021-09-03:
> When do you think they will be available for purchase?

@Danny2100 Shipping October, 1 and listed a couple of days before that

suburbiasystems, 2021-09-13:
I am interested too. I hope there is enough in stock Oct 1st.

j0rd, 2021-09-14:
I am interested in buying 2x of these E key to PCIe adaptors.

I'm an idiot, so I have a question about a potential product. Would it be possible to create a PCIe 4x device which can support 2x dual TPUs? Would it be possible to create something like this where you wouldn't need to use a PCIe hub/switch and provide direct access to all the TPU cores? Again I'm an idiot, but would like to know why this is (or is not) possible. Would it need bifurcation?

Again thanks for your hard work on this. I've been googling around attempting to find some device to hook this into an older dell r720 server and this is a life saver.

magic-blue-smoke, 2021-09-15:
> I'm an idiot, so I have a question about a potential product. Would it be possible to create a PCIe 4x device which can support 2x dual TPUs? Would it be possible to create something like this where [...]

Dual Edge TPU has two independent TPUs and for this reason two PCIe buses are required. PCIe x4 doesn't count as four buses, as it's four lanes, not four buses. I.e. most desktop PCIe slots can handle only one TPU (passively, without PCIe switch)

Without PCIe switch, its theoretically possible to use passive PCIe x16 to 4x m.2 NVMe BM key adapter with four Coral M.2 Accelerator B+M key cards. PC shall support PCIe bifurcation on this PCIe x16 slot: https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/2#issuecomment-904779681

I'm designing my adapters with PCIe switch onboard to use Coral Dual Edge m.2 cards in any system.

I do have plans to design adapters for 2-8 Dual TPUs, but can't promise any dates now

thePolarblair, 2021-09-18:
Definitely interested. Can you post a link to the sale page so I can bookmark it in anticipation of stock coming in?
Very cool idea. Assuming if I have two or even 3no. x1 slots spare I could put in three of your adapter boards and so support 3no. Coral dual edge tpus?

I do have a full size x16 slot available. I take it there is no off the shelf card from Amazon that would work support the dual edge tpu?

magic-blue-smoke, 2021-09-19:
> Definitely interested. Can you post a link to the sale page so I can bookmark it in anticipation of stock coming in?

Link will be posted here and on Twitter

> Very cool idea. Assuming if I have two or even 3no. x1 slots spare I could put in three of your adapter boards and so support 3no. Coral dual edge tpus?

I see no showstoppers to have several adapters and Dual Edge TPUs in one system

> I do have a full size x16 slot available. I take it there is no off the shelf card from Amazon that would work support the dual edge tpu?

To be fair, there is PCIe x16 card available from ASUS: https://store.asus.com/us/item/202103AM180000006/
I can't tell anything about this card now and I'll investigate more when I'm about to step into that segment.

light9, 2021-09-20:
Hello! I'm interested to get 2. Thanks a lot!

shtnarg, 2021-09-21:
I would Also Like to order 2.

tehniemer, 2021-09-23:
In for one

horspiste, 2021-09-23:
In for one

Jimmy-SafeCash, 2021-09-23:
I would like to purchase one of these.

irisc21464, 2021-09-24:
Hello! I would like to order four (4) low-profile PCIe x1 cards.

jrdean, 2021-09-25:
I could really use 1 of these. Thanks for your hard work and sharing such a useful board with the community.

j0rd, 2021-09-27:
Do you plan on selling this card with a low-profile PCIe bracket or standard bracket? Or will it be bracket-less? Just curious if I should be pre-ordering brackets.

magic-blue-smoke, 2021-09-28:
> Do you plan on selling this card with a low-profile PCIe bracket or standard bracket? Or will it be bracket-less? Just curious if I should be pre-ordering brackets.

@j0rd boards to be shipped with mounting screw for m.2 card and without bracket. Bracket mounting holes are according to PCIe specs, so if you have some lying around, they should fit.
Please order brackets after order confirmation, as demand is approaching number of boards in a first run

DrSpaldo, 2021-09-29:
Interested in 1 or 2

wills106, 2021-09-29:
Look forward to these becoming available!

Jimmy-SafeCash, 2021-10-01:
Any update on availability @magic-blue-smoke

Teme-V, 2021-10-03:
I'm interested of one adapter. Let's see if HP sff elitedesk support dual TPU using adapter. It didn't work at all in NIC e.key slot

lcasale, 2021-10-04:
hello, I would like a wow profile PCIe x1 please! Do you know when they will be available?

magic-blue-smoke, 2021-10-04:
@Jimmy-SafeCash @lcasale 
Boards are manufactured and tested.
There are some formal things that has to be done once and right
I do want to finish those this week and finally start sales

[104 more replies]