magic-blue-smoke/Dual-Edge-TPU-Adapter #5: Which m.2 variants for manufacturing?

magic-blue-smoke, 2021-08-21:
This issue is to separate desktop PCIe here /issues/4 and m.2 variants

Dual Edge TPU to m.2 adapter is possible in following configurations:
- m.2 A+E key 2242. This form-factor is non-standard (standard is 2230 for A/E keys) - you need to check clearance for longer card
- m.2 B+M key 2242, 2260 and 2280. 

Putting all of those into production is not easy, please let me know:
- which key and form factor you'd prefer
- quantity

wmchris, 2021-08-22:
i think the 2280 and the native x1 pcie version are the most favorable in this case. i dont see any market for a 2242 A+E card. i think a 2230 a+e with a riser would be more practicable.

magic-blue-smoke, 2021-08-23:
@wmchris now I have the same feeling, but before that I was thinking Coral Dual TPU B+M adapter should be most popular for Rock Pi's and other SBC with m.2 B key slot. I was wrong: most adapters went to PCIe slot with another COTS NVMe-to-PCIe adapter. Will see how it goes :)
Surely m.2 2280 B+M version seems not only more favorable, but also less constrained to manufacture

wmchris, 2021-08-23:
> I was wrong: most adapters went to PCIe slot with another COTS NVMe-to-PCIe adapter. Will see how it goes :)

well the reason is, that the native nvme slot supplies 4 pcie lanes, while your adapter shrink it to 1 pcie lane. 
in my case the pcie x1 slot on the mainboard is unused, as on most builds, while the nvme slots are occupied in the most cases

jeramiapoland, 2021-09-01:
I'm interested in a m.2 A+E key adapter at either length.

magic-blue-smoke, 2021-09-21:
m.2 B+M 2280 hand soldered prototype:

[image]

Teme-V, 2021-10-03:
I'm interested of b/m key adapter.
Hope can get dual TPU recognized with this in HP elitedesk as with onboard NIC m.2 e-key it didn't work - not even recognized

KingKoro, 2021-10-06:
I'm also interested in B+M Key Variant. 
I have a NanoPi M4v2 that has a daughter board for NVMe B+M Key with PCIe 2.1 x2 Lanes.
Does somebody know if there's a cheap e-key to b+m-key adapter that routes both pcie lanes through for dual edge tpu without any switch IC? Would it be difficult to design a simple passive adapter myself that routes both lanes?

Board: https://www.friendlyarm.com/index.php?route=product/product&product_id=268
NVMe Adapter: https://www.friendlyarm.com/index.php?route=product/product&product_id=262

krim404, 2021-10-06:
> Does somebody know if there's a cheap e-key to b+m-key adapter that routes both pcie lanes through for dual edge tpu without any switch IC? Would it be difficult to design a simple passive adapter myself that routes both lanes?

you cant simply create a passive adapter, as a general m-keyed m2 has only one pcie bus.

magic-blue-smoke, 2021-10-06:
> I'm also interested in B+M Key Variant. I have a NanoPi M4v2 that has a daughter board for NVMe B+M Key with PCIe 2.1 x2 Lanes. Does somebody know if there's a cheap e-key to b+m-key adapter that [...]

@KingKoro you can design semi-passive adapter only if SoC supports PCIe bifurcation, ie PCIe x2 has to be split into 2xPCIe x1 in hardware. Contact manufacturer to see if it’s possible.
Another issue is that there is only one clock line, while Dual Edge TPU requires two and you’d need clock fanout driver IC, that’s why it’s “semi-passive”

singsonn, 2021-10-06:
Hello, I'm also interested in B+M key (2280).

KingKoro, 2021-10-06:
> @KingKoro you can design semi-passive adapter only if SoC supports PCIe bifurcation, ie PCIe x2 has to be split into 2xPCIe x1 in hardware. Contact manufacturer to see if it’s possible. Another [...]

Thanks for the information, i didn't know that. According to [another thread](https://electronics.stackexchange.com/questions/449586/requirements-for-pcie-bifurcation-on-sbc) the RK3399 doesn't support PCIe bifurcation so that option is off the table for me then.
Still great work you're doing here, looking forward to the B+M Key Adapter becoming a real product.

jon-chen, 2021-10-11:
Interested in the B+M key, 2280. Thanks!

cdrose, 2021-10-12:
If the B+M key 2280 adapter does indeed work with a PCIe to 4 x M.2 adapter (with a suitable CPU and motherboard that support bifurcation), then this seems like a perfect solution for me. 8 x Coral TPUs at a fraction of the cost of the Asus CRL-G18U-P3D would be ideal for my application. I know you have said the 1 x PCIe adapter is the first to go to production, any ideas when the B+M 2280 might be available?

magic-blue-smoke, 2021-10-12:
> If the B+M key 2280 adapter does indeed work with a PCIe to 4 x M.2 adapter (with a suitable CPU and motherboard that support bifurcation), then this seems like a perfect solution for me. 8 x Coral [...]

@cdrose adapter shown here has tiny mistake that does not affect anything, but confuses when you know where to look at. I’d fix it and run small production run with availability around Christmas.

mike3e, 2021-10-12:
Hi!
I'm also interested in the B+M key (2280)

Mattias314, 2021-10-12:
I'd be interested a B+M key as well!

nborbit, 2021-10-13:
I am interested in B+M key (2280)

bipple294, 2021-10-16:
I am interested in two B+M key 2280s.

aaron2198, 2021-10-18:
I, as well as some friends that i'll get to post here, are also interested in some of these B+M key 2280's. +2 at least.

woolmonkey, 2021-10-30:
Interested in B+M key.  So the PCI-e x1 shows both with only 1 pci-e lane?

SecretiveShell, 2021-10-30:
I am expressing my interest as requested in the readme.

mbirrell66, 2021-11-01:
I'd be interested a B+M key as well!

badscr, 2021-11-01:
I’m interested. I’ve got 3 computers and 3 e-key dual tpu cards.

Each motherboard has One m.2 WLAN slot with A-Key.  And the Coral cards are E-Key.

Can I carefully cut the A-key slot (pin 8-16) into my Dual TPU Coral E-key card to make it an a+e card? Yah one tpu probably would not have pcie bus connection but 50% working is better then 0% also it would tell me if the motherboard has an pcie bus to the connector too.   My only concern is pin 66 (UIM_POWER_SNK) and 68 (UIM_POWER_SRC).

magic-blue-smoke, 2021-11-01:
> Can I carefully cut the A-key slot (pin 8-16) into my Dual TPU Coral E-key card to make it an a+e card?? Yah one tpu probably would not have pcie bus connection but 50% working is better then 0% [...]

@badscr your best result is getting only one core, as A key has only one PCIe bus and only E-key can have both. Worst case - not getting any if it's Intel CNVio rather than PCIe.

UIM_xxx pins are interface from m.2 card to SIM-card on motherboard. Obviously TPU card is not using those.

Writing this without looking into specs and this should not be taken as "go for dremel" :rofl:

FlorianZ, 2021-11-01:
Interested in B+M key (qty 1 or 2). Thank you!!!

ibrahimsoliman97, 2021-11-02:
I am interested in B+M Key, I got 2 Dual TPU Key E but I cannot run it on any of my PCs. 
Will a converter like the below work with Key E to Key M or at least detect a single TPU? 
[image]

https://shopee.com.my/M.2-A-E-Key-Slot-to-M.2-M-Key-Adapter-for-Intel-AX200-Bcm94352Z-Card-i.118950463.13926218941?sp_atk=66e6bfda-ab8e-43a6-9de1-abdc94275521

magic-blue-smoke, 2021-11-02:
@ibrahimsoliman97 I expect you'll get one TPU if m.2 slot is PCIe (not SATA)

ibrahimsoliman97, 2021-11-03:
@magic-blue-smoke Sure, I will give it a try. Thanks.

hack2spider, 2021-11-03:
B+M here 2 pcs`s possibly 4 ... to build to individual systems ... 

B+M in this scenario would use the dual tpu in full (both tpus) I presume

magic-blue-smoke, 2021-11-05:
This thread had intention to figure out which m.2 to produce.
m.2 2280 B+M version is planned for production with availability in January and I encourage everyone having this version in mind to apply for a waiting list here: https://mbs-shop.online/waitinglist
I'll be contacting people from the list prior to general sales

Calimerorulez, 2021-11-23:
> This thread had intention to figure out which m.2 to produce. m.2 2280 B+M version is planned for production with availability in January and I encourage everyone having this version in mind to [...]

@magic-blue-smoke 
The site for the waitinglist isn't reachable atm.

magic-blue-smoke, 2021-11-23:
@Calimerorulez it's slower to load than I'd expect, but working at the moment. Please let me know if problem persists

mbirrell66, 2021-11-24:
How about a PCIe card that could fit 2 x Dual M.2 Key E - that would be perfect for the Figate project. Maximum TPUs' supported is 4 (for some reason). Now is the time to stert pumping them out as you would have the Monopoly :)

ozett, 2021-11-24:
i recently thought of using multiple TPU in frigate, but its of no great benefit
https://github.com/blakeblackshear/frigate/issues/2281#issuecomment-974038686

if frigate is extended to run more AI-Models ([face-recognition was rumored](https://github.com/blakeblackshear/frigate/issues/260#issuecomment-968306156)) you may need one TPU for each Model..

magic-blue-smoke, 2021-11-24:
> How about a PCIe card that could fit 2 x Dual M.2 Key E - that would be perfect for the Figate project. Maximum TPUs' supported is 4 (for some reason). Now is the time to stert pumping them out as you would have the Monopoly :)

@mbirrell66 is that limit for Frigate? Or where it comes from?

kallejoken, 2021-11-27:
I have now got myself a new motherboard ASRock j3355b.  now I'm stuck with iommu_groups, the coral is in the same group as my network card and many other devices (group 5), can I somehow assign the coral to its own group?

mbirrell66, 2021-12-01:
@magic-blue-smoke That is something I have read in a review... somewhere :(. I have 6 Duel Corel TPUs' - When I have 6 of your boards I will put them all in my Frigate machine and let you know if I can use all 12 TPUs' at once!

ozett, 2021-12-01:
Multiple TPUs is only of small benefit in frigate right now, you could stay with only one.
or you put one AI-Model on each of the TPUs
https://github.com/blakeblackshear/frigate/issues/2281#issuecomment-974038686

mbirrell66, 2021-12-07:
@ozett yea you are right. I tried using cameras (Wyze) that didn't support ONVIF which was a mistake. I thought that my system couldn't handle 6 cameras. Once I replaced them with TP-Link 310s' the connection problems went away. I am still only using 1 TPU of my Dual and waiting to buy magic-blue-smoke's hardware solution.

Teme-V, 2022-01-07:
Any recent news about this Project? I've joined to waiting list, but probably bit too keen sitting down and wait

magic-blue-smoke, 2022-01-07:
> Any recent news about this Project? I've joined to waiting list, but probably bit too keen sitting down and wait

@Teme-V Current status is:
- got all parts and PCBs for January production run
- ordered parts and PCBs for February production run to dampen Chinese New Year holidays
- parts ordered for January and February production run combined, are to fulfill all current pre-orders
- got stencil printer, pick and place machine and reflow oven. These are entry level machines, making them work precisely enough to use 0402 parts is quite a learning experience 
- 40 m.2 boards were assembled today. Some SMDs got tombstoned, some pins have shorts, however nothing that can’t be fixed and general impression is optimistic
- will try PCIe x1 adapter assembly tomorrow 

All of this was done during holiday season and I’m happy with the balance I got between spending time with family and rushing around to get some parts, materials, arranging new lab and getting experience with machines and assembly process

jeramiapoland, 2022-01-23:
Is it sold out already? All I see is the 'out of stock'.

magic-blue-smoke, 2022-01-23:
> Is it sold out already? All I see is the 'out of stock'.

You’ve signed up for waiting list and should’ve received notification email today. Please use link in the email to purchase adapters.

jbrukardt, 2022-01-23:
waitlist the same as pre-order?   Would gladly pre-pay if that gets me in the order queue for pci-e 1x adapter.   It seems that by the time the email gets out... theyre already gone

magic-blue-smoke, 2022-01-23:
> waitlist the same as pre-order? Would gladly pre-pay if that gets me in the order queue for pci-e 1x adapter. It seems that by the time the email gets out... theyre already gone

I think I was using “pre-order” term a number of times instead of “waitlist” and the latter is more correct. Despite of I’ve invested significant amount of money for parts, equipment and lab, I don’t feel comfortable taking money before having product in my hands.
As I produce the boards, I’m sending emails with purchase link, ie 10 boards produced - emails for 10 boards are sent. People using purchase link should not see “out of stock”, however  technically it can be if payment failed, but CRM still keeps items booked

So if you’re in waiting list, make sure to check emails regularly including “spam” and “promotions” folders

mwheeler1982, 2022-01-23:
What email address or domain would this email come from? I’d like to
whitelist it so that it doesn’t go to spam.

jbrukardt, 2022-01-23:
> I think I was using “pre-order” term a number of times instead of “waitlist” and the latter is more correct. Despite of I’ve invested significant amount of money for parts, equipment and lab, I [...]

That sounds very fair.

ShadowKitty42, 2022-01-24:
Still waiting on my email, but I'm happy things are moving along smoothly.

magic-blue-smoke, 2022-02-02:
> What email address or domain would this email come from? I’d like to whitelist it so that it doesn’t go to spam.

@mwheeler1982 email sent

fryc88, 2022-03-02:
Still possible to sign up for E adapter?  For Wyse 5070 with M.2 E slot, lenght (for me personally) does not matter, but as short as possible.

ShadowKitty42, 2022-03-08:
Any updates on how this is going? Still haven't received my email yet, not sure I've missed it somewhere, but I hope not.

zbrozek, 2022-03-13:
I signed up once for personal projecting in... October? I think? And started to have a business interest for my day job and signed up again, but with my corporate email today or yesterday. Haven't heard back in either place, so I'm starting to wonder if there are other good solutions.

magic-blue-smoke, 2022-03-13:
> Any updates on how this is going? Still haven't received my email yet, not sure I've missed it somewhere, but I hope not.

@ShadowKitty42 with recent events I have to rebuild some of my processes. There are several good looking options and I'll be announcing updates soon in this thread, please follow: https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/8

> I signed up once for personal projecting in... October? I think? And started to have a business interest for my day job and signed up again, but with my corporate email today or yesterday. Haven't heard back in either place, so I'm starting to wonder if there are other good solutions.

@zbrozek all customers that made requests in October had notification letters. From your name, I see your requests from December and form yesterday and this is where line didn't get at the moment. To date, half of requests from from waiting list were fulfilled and I'd need about a month to resume sales with better availability, as well as payment and shipping options

zbrozek, 2022-03-13:
@magic-blue-smoke that's great to hear; thank you for the update!

ryanrhee, 2022-09-22:
Hi, the latest update in #8 says

> **m.2 B+M adapters** are expected to be available in ~3 weeks

which was posted >1mo ago. are there any updates regarding the b+m adapters?

magic-blue-smoke, 2022-09-25:
> Hi, the latest update in #8 says
> 
> > **m.2 B+M adapters** are expected to be available in ~3 weeks
> 
> which was posted >1mo ago. are there any updates regarding the b+m adapters?

Hi @ryanrhee 
B+M adapters are expected to be ready on September 30

zbrozek, 2022-09-25:
I'm certainly still interested in getting one.

magic-blue-smoke, 2022-09-25:
> I'm certainly still interested in getting one.

@zbrozek 
Consider subscribing [this thread](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/8) or following Twitter
Due to delayed demand I might not be touching waiting list on this batch

fryc88, 2022-09-26:
What about e key adapters?Will they ever be available?

zbrozek, 2022-09-26:
Ordering is not a FIFO queue based on the wait-list?

[33 more replies]