magic-blue-smoke/Dual-Edge-TPU-Adapter #6: SBC/Motherboards that support Dual M.2 / 2 buses

browntownington, 2021-09-15:
Firstly super kudos @magic-blue-smoke for developing these adapters.

I know the https://github.com/google-coral/edgetpu/issues/256 thread discusses support HW for Edge TPU but it's not really specific to the Dual Edge TPU or HW with 2 buses. OP notes M.2 dual tested with Nexcom.

Reading the comments there's lots of people getting one TPU working from the Dual. 

But it's difficult to determine what HW has 2 buses and could work with your adapters to support Dual.

Could we start a new thread here to track SBC's and Motherboard that's have been tested or untested (that support 2 buses) as the other thread is too vague.

magic-blue-smoke, 2021-09-15:
@browntownington I'm a little bit skeptical about tracking motherboards here, as original thread has more visitors.
Nevertheless, I don't have any objections to keep list of motherboards here. Feel free to edit own post, I also will be checking if there is something to add from posts below

**What to look at in specs**
- something like m.2 E-key **2x** PCIe x1
- m.2 connector pinout that has both PCIe buses: PCIe0 differential pairs at pins 35-45 and PCIe1 at pins 59-73

[image]

browntownington, 2021-09-17:
Thanks @magic-blue-smoke I have edited and reworded my OP.

Appreciate the guidance and things to look for. I still definitely find the key specific's and specifications extremely vague from manufacturers. It seems the bus specifics is often neglected and I am unsure if that just simply means it is 1x  PCIe. But I will attempt to dig deeeeeper