magic-blue-smoke/Dual-Edge-TPU-Adapter #9: share a m2 mkey to dual ekey

yike-citing, 2021-11-05:
Hello, your design is very interesting, but I want to design a passive switching solution by myself, which will transfer an M.2 Mkey plus a retimer to two ekeys, which will help me in the standard four-port plx There are 8 Coral Dual Edge TPUs implemented above. Can you share the packaging files of Coral Dual Edge TPU and ekey interface of your pcb design software with me? Thank you very much!

magic-blue-smoke, 2021-11-05:
@yike-citing 
Coral Dual Edge TPU pinout described in its datasheet and corresponds to m.2 electromechanical specification.
I have no idea how to implement your idea, ie passively without PCIe switch.

yike-citing, 2021-11-06:
I plan to use a retimer to generate four clock signals and use the pcie channel of pch. I adjust me to split the pch into four x1 pcie

yike-citing, 2021-11-06:
And the m2 mkey of most motherboards and notebooks are connected by pch

closed 2022-07-27 by magic-blue-smoke