magic-blue-smoke/Dual-Edge-TPU-Adapter #10: Pricing?

Makin-Things, 2021-11-08:
Hi, since you haven't got any for sale at them moment there is no mention of pricing anywhere. It would be nice if you published rough/estimated prices somewhere (including shipping costs).

magic-blue-smoke, 2021-11-08:
Hi @Makin-Things , prices are available at the store: https://mbs-shop.online
Having prices in Russian Ruble might confuse most of customers.
I was setting those to roughly match following prices:
Dual Edge TPU Adapter - PCIe x1 Low Profile ₽1800 ~ US$25
Dual Edge TPU Adapter - m.2 2280 B+M key ₽2100 ~ US$30
Shipping with Registered Airmail ~US$7
I'm looking forward adding faster and more predictable delivery options for the next production run

Makin-Things, 2021-11-08:
Sorry, I didn't see that. It wasn't obvious that there was stuff off the bottom of the screen. Closing now.

closed 2021-11-08 by Makin-Things