magic-blue-smoke/Dual-Edge-TPU-Adapter #11: increased latency?

vebmaster, 2021-12-10:
> One PCIe bus on m.2 E-key connector is not end of the World. With PCIe switch bus can be split in two busses at cost of slightly increased latency and shared bandwidth.

Hello.
1) What is the difference in milliseconds when using "Coral DUAL tpu" on PCIe x2 without an adapter and on PCIe x1 through an adapter?
2) Not sure why "Coral DUAL tpu" requires 2 PCIe lanes? Is the data bitrate really that big?

magic-blue-smoke, 2021-12-11:
HI @vebmaster 

> 1. What is the difference in milliseconds when using "Coral DUAL tpu" on PCIe x2 without an adapter and on PCIe x1 through an adapter?

This screenshot https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/3#issuecomment-885925888 tells that inference time increased from 7.5ms for single core to 8.5ms when TPUs behind PCIe switch. 

> 2. Not sure why "Coral DUAL tpu" requires 2 PCIe lanes? Is the data bitrate really that big?

Dual Edge TPU card requires two buses, not two lanes. If you take a look at picture of this card, you'll notice two cores, two sets of DC-DC converters and all another circuitry is duplicated. m.2 E-key slot implemented to full specs has two PCIe x1 buses, this makes two independent TPUs work. Not to be confused with number of lanes: you can attach NVMe drive to PCIe x1 slot at the price of 1/4th of bandwidth. Most m.2 E-key motherboards have only one PCIe bus and signals for one TPU are connected properly, for another just floating.

vebmaster, 2021-12-11:
> If you take a look at picture of this card, you'll notice two cores, two sets of DC-DC converters and all another circuitry is duplicated

Thanks a lot. Now I understood what was the matter.

Can you tell me what bitrate one TPU has?
Using a USB accelerator as an example:
> USB Accelerator datasheet: https://coral.ai/docs/accelerator/datasheet/
> 1. System requirements
> One available USB port (for the best performance, use a USB 3.0 port)

throughput USB 2.0 = up to 480 Mbps
USB 3.0 = up to 5Gbps
Is the data bitrate more than 480 Mbps?

magic-blue-smoke, 2021-12-11:
@vebmaster I don't have particular numbers and [this article](https://www.hackster.io/news/benchmarking-machine-learning-on-the-new-raspberry-pi-4-model-b-88db9304ce4) gives some impression on USB2.0/3.0 with Raspberry Pi4. Strange thing is that somehow RPi3 outperforms RPi4 using USB2.0

treichst, 2021-12-11:
> HI @vebmaster
> 
> > 1. What is the difference in milliseconds when using "Coral DUAL tpu" on PCIe x2 without an adapter and on PCIe x1 through an adapter?
> 
> This screenshot [#3 (comment)](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/3#issuecomment-885925888) tells that inference time increased from 7.5ms for single core to 8.5ms when TPUs behind PCIe switch.
> 
> > 2. Not sure why "Coral DUAL tpu" requires 2 PCIe lanes? Is the data bitrate really that big?
> 
> Dual Edge TPU card requires two buses, not two lanes. If you take a look at picture of this card, you'll notice two cores, two sets of DC-DC converters and all another circuitry is duplicated. m.2 E-key slot implemented to full specs has two PCIe x1 buses, this makes two independent TPUs work. Not to be confused with number of lanes: you can attach NVMe drive to PCIe x1 slot at the price of 1/4th of bandwidth. Most m.2 E-key motherboards have only one PCIe bus and signals for one TPU are connected properly, for another just floating.

I just got my dual TPU adapter in the mail.
I am getting 6.21 ms for a Single TPU and 7.91ms and 7.94ms for two Dual TPUs.

This is ran via docker-compose on an Ubuntu server virtual machine within the Proxmox hypervisor with a handful of other VMs running.

https://i.imgur.com/0RI5upr.png

magic-blue-smoke, 2021-12-11:
@treichst good to hear boards are arriving. Half of US orders had stops in Poland and Czech Republic before landing in JFK, yours was among these. That's crazy.

vebmaster, 2021-12-12:
> I am getting 6.21 ms for a Single TPU and 7.91ms and 7.94ms for two Dual TPUs.

Thx. 
Have you connected "dual TPU adapter" directly to m.2 slot or via multiplexed adapter?

What frequency does your TPU run at? 125 MHz / 250 MHz / 500 MHz?
https://coral.ai/docs/m2-dual-edgetpu/datasheet/
3.2 Power consumption -> Table 3

vebmaster, 2021-12-15:
> If you take a look at picture of this card, you'll notice two cores, two sets of DC-DC converters and all another circuitry is duplicated.

According to the PCIe specification, the device can draw up to 3A at 3.3v.
According to spec, one coral TPU makes short-term surges up to 3A.

https://coral.ai/docs/m2-dual-edgetpu/datasheet/
https://coral.ai/docs/m2/datasheet/
3.2 Power consumption
> Although the average current drawn from the 3.3V supply by each Edge TPU is typically less than 500 mA, brief current transients that occur during inferencing can reach roughly 3 A.

In case of using two TPUs, is it possible that there is a shortage of power?

magic-blue-smoke, 2021-12-17:
> In case of using two TPUs, is it possible that there is a shortage of power?

@vebmaster It mostly depends on how power lines are routed on motherboards. To help this, there are ceramic capacitors placed close to m.2 connector

vebmaster, 2021-12-23:
Я правильно понимаю, что Вас Александр зовут? Не против если я по русски продублирую, чтоб правильней поняли меня. А то я по русски бывает еле объясняюсь правильно, не то что по английски.
Я не совсем в деталях понимаю все тонкости PCIe и т.д., поэтому пишу как есть, как я понимаю.
> В основном это зависит от того, как на материнских платах проложены линии питания.

А какая разница как на материнских платах проложены линии питания, если по спецификации на PCIe линию выдаётся только 3A при 3.3v максимум. А у нас два TPU (две PCIe линии) в одной PCIe линии (как розетка).
И в случае одновременного скачка на обоих TPU до 3А получится 6А при 3.3v. Соответственно PCIe не даст 6А, а даст только 3А.
Поэтому и возникл дополнительный вопрос:
1) Возможен ли дифицит мощности?
2) насколько это вообще вероятно - одновременный скачёк до 3А на обоих TPU.
или что то недопонимаю и нет поводов для беспокойства?

Александр, а вы пробовали долгое время, несколько дней хотябы, работать на таком переходнике с dual TPU с моделями по детекции объектов/лиц? Не было проблем?

----
Do I understand correctly that your name is Alexander? Do not mind if I duplicate it in Russian so that they understand me more correctly. Otherwise, I can hardly explain myself correctly in Russian, let alone in English.
I do not quite understand all the details of PCIe, etc., so I write it as it is, as I understand it.
> t mostly depends on how power lines are routed on motherboards

And what's the difference how the power lines are laid on the motherboards, if according to the specification for the PCIe line, only 3A is issued at 3.3v maximum. And we have two TPUs (two PCIe lanes) in one PCIe lane (like a socket).
And in the case of a simultaneous jump on both TPUs to 3A, you get 6A at 3.3v. Accordingly, PCIe will not give 6A, but only 3A.
Therefore, an additional question arose:
1) Is there a power shortage?
2) how likely it is at all - a simultaneous jump to 3A on both TPUs.
or something I misunderstand and there is no reason for concern?

Alexander, have you tried for a long time, at least a few days, to work on such an adapter with dual TPU with models for object / face detection? There were no problems?

NickM-27, 2022-07-20:
Just got my PCIe adapter in, I see no difference in latency from before running just one

vebmaster, 2022-07-20:
> Just got my PCIe adapter in, I see no difference in latency from before running just one

Thank you.
And what is your latency through the PCIe adapter?

NickM-27, 2022-07-20:
> > Just got my PCIe adapter in, I see no difference in latency from before running just one
> 
> Thank you. And what is your latency through the PCIe adapter?

I've added sensors in frigate to track their latency https://github.com/blakeblackshear/frigate-hass-integration/pull/313 as I am using them for frigate, in my case both TPUs stay within 6.5 and 6.8 ms of latency

vebmaster, 2022-07-22:
> in my case both TPUs stay within 6.5 and 6.8 ms of latency

Thx. What is the frequency of both TPUs?
Low operating frequency - 125 MHz
Reduced operating frequency - 250 MHz 
Max operating frequency - 500 MHz

Are you using TPU cooling? Radiators, fans?

NickM-27, 2022-07-22:
> Thx. What is the frequency of both TPUs? Low operating frequency - 125 MHz Reduced operating frequency - 250 MHz Max operating frequency - 500 MHz
> 
> Are you using TPU cooling? Radiators, fans?

Running at max 500 MHz. My NAS case is very well ventilated but it has no dedicated heat sink or fans. One TPU sits at 55 C while the other is at 63 C

NickM-27, 2022-07-22:
> Have you tried for a long time, at least a few days, to work on such an adapter with dual TPU with models for object / face detection? There were no problems?

Yes, it's been running for 3 days running frigate and graphing to HomeAssistant and they have stayed at 6.5 ms and the temps I mentioned previously

[image]

[image]

vebmaster, 2022-07-23:
> Running at max 500 MHz

Thank you. What device do you have TPU connected to?
What model are you using?

NickM-27, 2022-07-23:
> > Running at max 500 MHz
> 
> 
> 
> Thank you. What device do you have TPU connected to?
> 
> What neural model are you using?

Using this repos dual pcie adapter, ASUS b550m motherboard. 

Using https://github.com/blakeblackshear/frigate/blob/37325c70bad38b7b4db5846a43c293cca10ae5a5/docker/Dockerfile#L114

vebmaster, 2022-07-28:
@NickM-27, thanks, you helped me a lot.