magic-blue-smoke/Dual-Edge-TPU-Adapter #12: Flexible extension cable for m.2

vebmaster, 2021-12-12:
Hello.
I plan to connect the "coral dual tpu" via your adapter to the "ROCK 3A" single board (https://wiki.radxa.com/Rock3/hardware/3a) in the M.2 E-key slot. There is little space, only for 22x30 boards.
It is logical to connect the adapter via a flexible m.2 extension.
Please tell me which flexible extension cord is right for me?

magic-blue-smoke, 2021-12-12:
@vebmaster E-key Dual Edge TPU adapter was prototyped https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter#m2-ae-version however it's not in production due to no demand. Additionally, it has 2242 size and from what I see it will conflict with GPIO pins on your SBC. You can take a look at cables available from Chinese suppliers, there can be some options, for example E-key to desktop PCIe 

For mentioned SATA HAT there should be adapter and cable available from manufacturer. Something like this https://wiki.radxa.com/Rockpi4/hardware/M2_extend but I'm not sure about compatibility with your model. Isn't it included with SATA HAT?

vebmaster, 2021-12-13:
@magic-blue-smoke 
> You can take a look at cables available from Chinese suppliers, there can be some options, for example E-key to desktop PCIe

Is this an option suitable?
N1AC option (color) - M.2 A-E key NGFF to PCIe x1 - https://aliexpress.ru/item/1005003077375436.html
[image]
I insert into the PCIe x1 slot [Dual Edge TPU Adapter - PCIe x1 Low Profile](https://mbs-shop.online/tproduct/365624859-195484769511-dual-edge-tpu-adapter-pcie-x1-low-profil)

So it is possible?

> For mentioned SATA HAT there should be adapter and cable available from manufacturer.

Yes, there is a cable in the kit, an example of connecting "Penta SATA HAT" to "Rock pi 4" is shown here - https://hardware.developpez.com/dossiers/NAS/RockPi-3-Penta-Sata-Hat/

The "Penta SATA HAT" is also compatible with the "Rock 3a". The manufacturer says that before buying it is necessary to clarify whether the new revision of "Penta SATA HAT"? - https://forum.radxa.com/t/penta-sata-hat-supported-for-rock-3a/7998/2

magic-blue-smoke, 2021-12-13:
> Is this an option suitable? N1AC option (color) - M.2 A-E key NGFF to PCIe x1 - https://aliexpress.ru/item/1005003077375436.html I insert into the PCIe x1 slot [Dual Edge TPU Adapter - PCIe x1 Low [...]

I meant these type of cables. Idea and cable seem ok, however I can't guarantee it will work. One of the questions: are power cables good enough? You'd need good 3.3V power supply wires in this cable. 

> Yes, there is a cable in the kit, an example of connecting "Penta SATA HAT" to "Rock pi 4" is shown here - https://hardware.developpez.com/dossiers/NAS/RockPi-3-Penta-Sata-Hat/
> The "Penta SATA HAT" [...]

I can't really comment on 3rd party products without having real experience with it. However your config seems really interesting, please share results if you succeed with this.

vebmaster, 2021-12-13:
> I meant these type of cables. Idea and cable seem ok, however I can't guarantee it will work. One of the questions: are power cables good enough? You'd need good 3.3V power supply wires in this cable.

In the descriptions and other pictures, an additional power cable is included. - https://aliexpress.ru/item/1005003068741226.html
```SATA power adapter cable is included and must be connected to a 12V power source```
[image]

Do I need this extra power for the "Coral dual tpu" board?
Or is 3.3V flex cable enough?

vebmaster, 2021-12-13:
3.3v
[image]

vebmaster, 2021-12-13:
Presumably such an adapter should work too?
M.2 А+Е key to M.2 B+M - https://www.amazon.com/Sintech-NGFF-NVME-WiFi-Cable/dp/B07DZF1W55/

magic-blue-smoke, 2021-12-17:
> In the descriptions and other pictures, an additional power cable is included. - https://aliexpress.ru/item/1005003068741226.html `SATA power adapter cable is included and must be connected to a 12V power source` 

> Do I need this extra power for the "Coral dual tpu" board? Or is 3.3V flex cable enough?

@vebmaster 
This power cable has 12V (yellow) and 5V (red). 
12V is only used for fan connector on adapter
5V is not used by PCIe slot (by PCIe specs)
It means that this power cable will not help with better power provisioning to Coral card and stability depends on how good flex is, what current Rock3A can supply and how good it is to transient currents.