magic-blue-smoke/Dual-Edge-TPU-Adapter #13: PCIe x16 adapter board

valioiv, 2022-01-14:
Hey @magic-blue-smoke , are able to design and produce a board similar to ASUS one:
https://iot.asus.com/news/5nkjk29i/

It is tooo expensive and I'm pretty sure it can be build to times lower price!

magic-blue-smoke, 2022-01-14:
@valioiv I keep in mind going this way one day, but can't even tell if it can be made this year. 
Theoretically you can have cheap x4 dual TPUs with 4 of my m.2 BM adapters and off the shelf PCIe x16 to 4x NVMe board. Note that for the most of that NVMe adapter boards you'll need CPU to support PCIe bifurcation.

valioiv, 2022-01-17:
> @valioiv I keep in mind going this way one day, but can't even tell if it can be made this year. Theoretically you can have cheap x4 dual TPUs with 4 of my m.2 BM adapters and off the shelf PCIe x16 [...]

I got it! Thanks! Wish you luck and keep us posted for such a project!
BTW, I filed myself in the waiting list for a PCIe adapter board for Dual TPU. Hope you'll fulfill my wish :)