magic-blue-smoke/Dual-Edge-TPU-Adapter #14: Buy now not visible

Makin-Things, 2022-01-23:
Hi,

I saw your comment in https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/8 but for me it still says coming soon, not buy now.

Hopefully they haven't all sold already!?

Simon

magic-blue-smoke, 2022-01-23:
> Hi,
> 
> I saw your comment in #8 but for me it still says coming soon, not buy now.
> 
> Hopefully they haven't all sold already!?
> 
> Simon

@Makin-Things 
I’m going through waiting list at the pace of boards production and sending link to purchase page. Initially, on that page “buy now” button was not working for a first hour or so and then fixed.
If you’ve received boards availability email, please use attached link instead of store front page. If not, sorry for confusion.

Makin-Things, 2022-01-23:
Thanks, closing now.

closed 2022-01-23 by Makin-Things