magic-blue-smoke/Dual-Edge-TPU-Adapter #16: SSD and dual TPU on the same adapter

fantasmisiciliani, 2022-02-19:
Hello Magic Blue Smoke, I read you're planning to use more lanes for the upstream port of the PCIe bridge.
Do you think it's possible to make a B+M key adapter (22110 format?) to host an M key 2242 SSD (at least with 2 lanes) and the Coral dual edge TPU using a PCIe Gen 3 switch (on the backside?), like the Asmedia ASM2812I or the Diodes PI7C9X3G808GP? I have to say that the first one is even not listed on distributors' catalogue and the second one is distributed only by Future Electronics with no availability at the moment. Both are in BGA package.
I know they cost 3-4 times the price of the ASM1184E, but the idea is to mantain the NVME SSD in SFF or USFF computers and add the Dual TPU, with a good bandwidth for both devices.
They could share a single heatsink and have an additional power input header if needed.
Somebody interested in such adapter, like me, could contribute for the prototypes, if you think you can realize it.

magic-blue-smoke, 2022-02-19:
Hi @fantasmisiciliani 
This type of adapter is possible from electronics/schematics point of view, however I see this case as very specific. I'd need demand for such adapters to be clearly identified to design it. Current plans are more towards having more TPUs.

fantasmisiciliani, 2022-02-19:
Yes, I fully understand. I hope some other has the same need and read this issue.
In the meanwhile I'm going to apply to the wishlist. I'm only waiting to receive the mini pc I ordered (Lenovo M70q) to know if its E-key slot is CNVIO2 or PCIe (the specs include both Intel AX201 and Realtek 8822CE), and in the case if single or double bus.
Thanks

magic-blue-smoke, 2022-02-19:
> Yes, I fully understand. I hope some other has the same need and read this issue.

Sure, I’ll keep this issue open in case someone else needs this type of adapter. Need to say going down to Gen 2 would make it much easier

> In the meanwhile I'm going to sign up for the waitlist. I'm only waiting to receive the mini pc I ordered (Lenovo M70q) to know if its E-key slot is CNVIO2 or PCIe (the specs include both Intel AX201 and Realtek 8822CE), and in the case if single or double bus (highly improbable).
> Thanks

It’s too sad manufacturers only list compatible components, rather than specs and pinouts

fantasmisiciliani, 2022-02-19:
> Sure, I’ll keep this issue open in case someone else needs this type of adapter. Need to say going down to Gen 2 would make it much easier
Thank you.
Of course, much easier and cheaper. I was thinking about Gen3 to get best performances also from the disk, whose bandwidth would be halved by lanes splitting, considering both the SSD and the slot at 4x. Using Gen2 it would be halved twice, down to about 1 GB/s, that's however not bad even for a small server.
> It’s too sad manufacturers only list compatible components, rather than specs and pinouts
I agree. Today technical knowledge is spreading fast, so manufacturers are trying to limit in every way our chances to get something "extra" from their packed -almost armoured- solutions.

magic-blue-smoke, 2022-05-26:
@fantasmisiciliani I have an update here: magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/8

fantasmisiciliani, 2022-08-05:
Hello @magic-blue-smoke as you wrote, b+m adapter should be available next month. will you follow waiting lists?

magic-blue-smoke, 2022-08-16:
> Hello @magic-blue-smoke as you wrote, b+m adapter should be available next month. will you follow waiting lists?

Hi @fantasmisiciliani 

Sorry for late reply, it's summer and sun time, while the boards are in production :)

B+M adapters are planned for September as noted in [Announcements thread](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/8#issuecomment-1197505081). Can be a small slip there, but still September.

For notifications and updates, following on Twitter or subscribing Announcement thread are the best options. Notifications using waiting list is something I'm working on now and expect it to be ready by B+M adapter launch.

fantasmisiciliani, 2022-08-20:
Yes, I've already subscribed Announcements thread ;)

Tweak4141, 2023-11-24:
Quite old, but I'd like to see something like this as well.