magic-blue-smoke/Dual-Edge-TPU-Adapter #17: PCIe x1 vs PCIe x16 vs M.2 SATA

TheStigh, 2022-02-21:
Hi @magic-blue-smoke 

Got my adapters from you today! THANK YOU!

Been some years since I played with PC and PCI specs.
I got the [](https://www.asus.com/Motherboards-Components/Motherboards/CSM/PRIME-H570M-PLUS-CSM/techspec/)
I use Intel 11th Gen CPU, and my M.2_1 are occupied with a NVMe drive.

Could you please tell me;

- Can I use the `PCIe x1` in the `PCIe x16` slot and get the same performance?
- Can I use the M.2 adapter in the M2_2 slot, listed as: `M.2_2 slot (Key M), type 2242/2260/2280 (supports PCIe 3.0 x4 & SATA modes)`

On my motherboard, I'm not sure if I can use PCIEX16_1 (should be able as I use Intel 11th gen).

Look forward for your reply :)

magic-blue-smoke, 2022-02-21:
> Hi @magic-blue-smoke
> 
> Got my adapters from you today! THANK YOU!

Glad to hear they arrived. DHL was quite fast, sad they didn’t make it on Friday 

> Been some years since I played with PC and PCI specs. I got the I use Intel 11th Gen CPU, and my M.2_1 are occupied with a NVMe drive.
> Could you please tell me;
> * Can I use the `PCIe x1 adapter` in the `PCIe x16` slot and get the same performance?

PCIe x1 cards should work in x16 slots without any issues

> * Can I use the M.2 adapter in the M2_2 slot, listed as: `M.2_2 slot (Key M), type 2242/2260/2280 (supports PCIe 3.0 x4 & SATA modes)`

m.2 card has ID pins to tell the host if it’s expecting SATA or PCIe interface. I set those to ask system to provide PCIe, so it should be fine

TheStigh, 2022-02-21:
Thx bud, I'll shut down the system tonight or tomorrow and do some installs.

Btw; ref the motherboard I linked; do you think I can use the PCIEX16_1 also?

magic-blue-smoke, 2022-02-21:
I can make own tests next week regarding using adapter in x16 slot. 
I didn't catch what's your MB model, however from what I see and from PCIe specs, there should be no issue using my adapter in PCIe x16 slot

TheStigh, 2022-02-21:
>I didn't catch what's your MB model, however from what I see and from PCIe specs, there should be no issue using my adapter in PCIe x16 slot

Sorry, here it is: https://www.asus.com/Motherboards-Components/Motherboards/CSM/PRIME-H570M-PLUS-CSM/techspec/

magic-blue-smoke, 2022-02-22:
Looked up the specs and:
- PCIe x1 adapter can be used in any PCIe slot
- m.2 BM adapter can be used in M.2_1 slot with 11th Gen CPU (which is your case) and M.2_2 slot (this one is PCIe/SATA and I set ID pins on my adapter to request PCIe from the system)

TheStigh, 2022-02-22:
@magic-blue-smoke 

```
stigh@ubuntu:~$ lspci -nnk | grep -i tpu
04:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
        Subsystem: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
05:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
        Subsystem: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
07:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
        Subsystem: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
0a:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a] (rev ff)
0b:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
        Subsystem: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
```

1 * Dual TPU in M.2 SATA
1 * Dual TPU in PCIe x16
1 * SIngle TPU Mini-PCIe to PCIe adapter in PCIe x1

:)

Though, I have issues connecting the single TPU to Frigate...

pdecat, 2022-02-22:
> Though, I have issues connecting the single TPU to Frigate...

Are you are using the Frigate Home Assistant add-on? If so, it only requests access to the first 3 APEX devices:

https://github.com/blakeblackshear/frigate-hass-addons/blob/c4b8bbf7100efdc95c659c4430056d6afab6fce1/frigate/config.yaml#L25-L27

But then, not sure how you would manage to get 4 TPUs working :confused:

TheStigh, 2022-02-22:
> Are you are using the Frigate Home Assistant add-on? If so, it only requests access to the first 3 APEX devices: [...]

No, running standalone in Docker on Ubuntu.

closed 2022-02-25 by TheStigh