magic-blue-smoke/Dual-Edge-TPU-Adapter #19: PCI Bracket for x1 adapter

coagentpai, 2022-03-08:
Does anyone know of a PCI (metal or otherwise) slot bracket that fits the adapter well?  I am hoping to be able to screw it in to the case.

I tried:

[Keystone 9203](https://www.digikey.com/en/products/detail/keystone-electronics/9203/317297)

But the mounting holes in the board don't line up. I seem to remember seeing a suggestion on here for one but I can't seem to track it down again.

magic-blue-smoke, 2022-03-08:
These two seem fit:
https://www.amazon.com/Standard-Height-Bracket-Profile-Cards/dp/B00BF7Q4MU/
https://www.amazon.com/Standard-Full-Height-Bracket-9265-8i/dp/B00HWH9M4Y/

The keywords are "Full Height Bracket for Low Profile Card" and distance between mounting tabs should be 54mm

taiheng, 2023-07-14:
These items have the correct spacing on Aliexpress

https://www.aliexpress.com/item/1005004003916975.html?spm=a2g0o.order_list.order_list_main.5.64341802ZRJ2te
https://www.aliexpress.com/item/1005004003916975.html?spm=a2g0o.order_list.order_list_main.4.64341802ZRJ2te