magic-blue-smoke/Dual-Edge-TPU-Adapter #20: Will source be available at some point?

zbrozek, 2022-03-12:
I've been hoping for availability of boards for quite a long time and at this point would be willing to invest the work to source and assemble them myself. Will the source files (schematic and layouts) be available at some point?

baskwo, 2022-03-12:
I would also be interested.
I live in Canada and if anything maybe I could check how hard it is to produce them near me and maybe be a distributor?
I would need to do some research first of course.

magic-blue-smoke, 2022-03-12:
@zbrozek @baskwo 
I don't plan to release source files. Now I'm negotiating to have distribution and/or production within EU. Canada seems interesting in terms of distribution. If interested in distribution, please leave a message at the bottom of the page here: https://mbs-shop.online/

closed 2022-07-27 by magic-blue-smoke