magic-blue-smoke/Dual-Edge-TPU-Adapter #21: PCIe x4 to four m.2 E-key slots adapter

gmpreussner, 2022-03-15:
Hello,

On the waitlist I saw on option for "PCIe x4 to four m.2 E-key slots adapter".

Will this adapter support 4 x Dual TPUs for a total of 8 TPUs?
Do you have a timeline for when, approximately, this may be available?

Thank you so much for your efforts!

magic-blue-smoke, 2022-03-15:
Hi @gmpreussner , 

I originally had idea to make 4x Dual TPU for 8 TPUs total and now this one is on hold
However 2x Dual TPU for 4 TPUs is planned for 2022Q3

RedAnon, 2023-01-03:
Hi, any news on that 2x Dual TPU adapter?
Right now I have two "Low profile PCIe x1 version" adapter but I would like to be able to free one of the used slots on my motherboard.

If it was scrapped for some reason would it be possible to use some of the AliExpress adapters out there?

magic-blue-smoke, 2023-01-05:
@RedAnon 
> Hi, any news on that 2x Dual TPU adapter? Right now I have two "Low profile PCIe x1 version" adapter but I would like to be able to free one of the used slots on my motherboard.
> 
> If it was scrapped for some reason would it be possible to use some of the AliExpress adapters out there?

Project is not scrapped. It's going slower than I'd like it to go, however current state is: prototype PCBs and parts are ordered and I expect to have it assembled in a few weeks

JackBeQuick87, 2024-05-22:
Hello and thank you for your work. 
Can you post an update on the status of the pcie x4 adapter?

magic-blue-smoke, 2024-05-27:
Hi @JackBeQuick87 
The first prototype I got had compatibility issues, so I had to find another PCIe switch IC and redesign the board completely. 
Current state is:
- redesigned prototype is being manufactured at the moment, as well as heatsink 
- sourcing accessories

I expect to receive it in 3 weeks and have some more news then

SrPatinhas, 2024-09-10:
Hey @magic-blue-smoke, thanks for this awesome components!

Any idea for the ETA of being available the PCIe x4? and what would be the price +/-?

magic-blue-smoke, 2024-09-11:
Hi @SrPatinhas 
I've got a new prototype and see some strange results. I'm using `multiple_tpus_performance_analysis.py` to test TPUs and for some reason `ssd_mobilenet_v1_coco_quant_postprocess_edgetpu.tflite` and `ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite` models just hang entire system, while others are running just fine
Going to try another PC and tweak some settings, so I'm not at the point where I can make solid statements on availability

magic-blue-smoke, 2024-09-17:
Ok, that's promising
```
============Summary==========
---------------------------
Model: mobilenet_v1_1.0_224_quant_edgetpu.tflite
Single Edge TPU base time: 108.442818 seconds
# TPUs: 2, speedup: 1.954415
# TPUs: 3, speedup: 2.864792
# TPUs: 4, speedup: 3.160298
---------------------------
Model: mobilenet_v2_1.0_224_quant_edgetpu.tflite
Single Edge TPU base time: 115.072573 seconds
# TPUs: 2, speedup: 1.981434
# TPUs: 3, speedup: 2.893779
# TPUs: 4, speedup: 3.351011
---------------------------
Model: ssd_mobilenet_v1_coco_quant_postprocess_edgetpu.tflite
Single Edge TPU base time: 174.176146 seconds
# TPUs: 2, speedup: 1.990360
# TPUs: 3, speedup: 3.002060
# TPUs: 4, speedup: 4.018068
---------------------------
Model: ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite
Single Edge TPU base time: 199.211695 seconds
# TPUs: 2, speedup: 1.988264
# TPUs: 3, speedup: 3.040751
# TPUs: 4, speedup: 4.012230
---------------------------
Model: inception_v1_224_quant_edgetpu.tflite
Single Edge TPU base time: 169.230412 seconds
# TPUs: 2, speedup: 2.086801
# TPUs: 3, speedup: 3.097807
# TPUs: 4, speedup: 4.062363
---------------------------
Model: inception_v2_224_quant_edgetpu.tflite
Single Edge TPU base time: 532.101504 seconds
# TPUs: 2, speedup: 2.005481
# TPUs: 3, speedup: 3.005257
# TPUs: 4, speedup: 3.987482
---------------------------
Model: inception_v3_299_quant_edgetpu.tflite
Single Edge TPU base time: 1560.548208 seconds
# TPUs: 2, speedup: 2.003140
# TPUs: 3, speedup: 3.007169
# TPUs: 4, speedup: 4.007953
---------------------------
Model: inception_v4_299_quant_edgetpu.tflite
Single Edge TPU base time: 3049.916455 seconds
# TPUs: 2, speedup: 2.003037
# TPUs: 3, speedup: 3.007209
# TPUs: 4, speedup: 4.007852
```

SrPatinhas, 2025-05-06:
Any news regarding this?