magic-blue-smoke/Dual-Edge-TPU-Adapter #22: Next shipment?

tempaone, 2022-03-15:
Hello,
When will the next shipment be available as it states all sold out and go on waiting list but no estimated time it will be available again.
Cheers

magic-blue-smoke, 2022-03-15:
Hi. Boards are actually in stock, however due to recent changes selling those makes sense from 10pcs.
I'm working on several alternative options to have stable availability and no MOQ; I expect to have it ready for shipment in 4-6 weeks

cegit55, 2022-03-22:
Thanks for the update - I am eagerly awaiting an m.2 B+M!

steppel, 2022-04-13:
So how to order with MOQ 10 .. the page still shows out of stock

cegit55, 2022-05-26:
Hi there @magic-blue-smoke,
Do you have any idea when the m.2 B+M adapter will become available?
I've been waiting (semi)patiently since the end of February.

magic-blue-smoke, 2022-05-26:
@cegit55 @steppel lot of things changed, update is available here: magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/8

magic-blue-smoke, 2022-10-11:
m.2 2280 BM adapters are available for purchase now: https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/8#issuecomment-1273867495

closed 2022-10-11 by magic-blue-smoke