magic-blue-smoke/Dual-Edge-TPU-Adapter #23: Verifying before ordering TPUs

phabibjr, 2022-04-14:
Since all of The Coral TPUs are on backorder, I’m going to go ahead and order them now with the hope of getting them by the end of the year. I’m just wanting to verify that you’re still planning to release another round of the dual TPU to PCIe adapters before placing an order on either the dual or PCIe version TPUs. 

Also, I would be very interested in the 2x dual TPU to PCIe adapter if you still plan to release that later this year.  Would you happen to know if individual TPUs could be assigned to different virtual machines (like in Proxmox) using your adapter?

Thank you.

magic-blue-smoke, 2022-05-26:
@phabibjr my current plan is to manufacture and ship the boards from China. Optimistically B+M adapters will be available in ~2 months, you can follow this thread: magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/8

magic-blue-smoke, 2022-07-15:
@phabibjr TPU to PCIe adapters are available now:
https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/8#issuecomment-1181686907

If not in stock by the time you're reading this - don't worry, I'll run another batch. 
For the news and boards availability consider subscribing GitHub thread above or follow me on Twitter: [@magic__smoke](https://twitter.com/magic__smoke)

magic-blue-smoke, 2022-07-27:
@phabibjr first batch manufactured and sold from China is sold out, however another production run for PCIe adapters is started and re-stock is expected  in 3 weeks.

I encourage you to subscribe [Store update issue](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/8) or follow updates on [Twitter](https://twitter.com/magic__smoke)

closed 2022-07-27 by magic-blue-smoke