magic-blue-smoke/Dual-Edge-TPU-Adapter #25: Only one tpu showing up with PCIe adapter

foxy82, 2022-07-18:
Hi. 

I have just installed the PCIe adapter with the coral dual tpu board. However I'm still only seeing a single Coral TPU. I followed the instructions on the coral page to install the Linux drivers. Is there anything else I need to do to see both TPUs?

magic-blue-smoke, 2022-07-18:
Hi @foxy82 
On host system you should see two instances of Coral TPUs with 'lspci' no matter if drivers are installed or not.

When drivers are properly installed, no additional steps are needed to see both apex0 and apex1 devices.

To use TPUs with Virtual Machine, configuraton needed for PCIe passthroug for both TPUs

When using Frigate, both TPUs should be configured in 'detectors' section

Are you using TPU within VM or without?
Also, could you show output of `lspci`?

magic-blue-smoke, 2022-07-18:
From what I see here, the issue is with hardware. `lspci` should show two `System peripheral: Global Unichip Corp. Coral Edge TPU` lines, because TPUs are independent 

Could you please try following:

- Remove TPU card from adapter
- Clean contacts on m.2 card
- Inspect TPU card for mechanical damage: TPUs and PMICs are bare flip chips and easy to break
- Reinsert card and try again with `lspci`
- If have another TPU card (or adapter) try different combinations of those

NickM-27, 2022-07-18:
@magic-blue-smoke I believe I should be getting my PCIe adapter tomorrow (very fast shipping 👌 ) so I haven't tested yet. 

One question I had that could potentially be affecting OP as well, does it matter the BIOS settings for the PCIe lanes? For example my mobo defaults to Gen 3 but I read that this device utilizes Gen 2, could that cause some issue?

I know PCIe is supposed to be backwards compatible but after hearing issues of gen 4 GPUs not showing up when using incorrect BIOS settings, figured I would ask.

baskwo, 2022-07-18:
Personally I received the adapter few days ago and I got no issue at all. Plugged it into my proxmox server, pci passthrough into a HomeAssistant OS VM. Installed Frigate add-on for HomeAssistant, configured it and it works well :)

baskwo, 2022-07-19:
@foxy82 Something to try is to make sure you have iommu isolation. With my AMD 1700x I had to tweak few things in the kernel so it could passthrough correctly. Maybe it's what made the adapter had no issue

magic-blue-smoke, 2022-07-19:
> @magic-blue-smoke I believe I should be getting my PCIe adapter tomorrow (very fast shipping ok_hand ) so I haven't tested yet.
> 
> One question I had that could potentially be affecting OP as well, does it matter the BIOS settings for the PCIe lanes? For example my mobo defaults to Gen 3 but I read that this device utilizes Gen 2, could that cause some issue if it isn't set to Gen 2?
> 
> I know PCIe is supposed to be backwards compatible but after hearing issues of gen 4 GPUs not showing up when using incorrect BIOS settings, figured I would ask.

@NickM-27 I've tested adapter on a number of motherboards with Gen3 x4 PCIe slot and it falls back to Gen2 x1 without any issues, so I don't expect problems here

> Personally I received the adapter few days ago and I got no issue at all. Plugged it into my proxmox server, pci passthrough into a HomeAssistant OS VM. Installed Frigate add-on for HomeAssistant, configured it and it works well :)

@baskwo Glad to hear boards are arriving!

foxy82, 2022-07-19:
> From what I see here, the issue is with hardware. `lspci` should show two `System peripheral: Global Unichip Corp. Coral Edge TPU` lines, because TPUs are independent
>
> Could you please try [...]

Hi - so I removed the adapter looked it over and it looks ok. I checked the coral board and it also looks ok. Cleaned both and re-inserted them and got the same outcome. I don't have another dual coral and I don't have another dual TPU adapter board. I do still have the original single lane adapter board I was using (https://www.amazon.co.uk/gp/product/B09F64TJ1W/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

I tried the dual adapter board in another computer and get the same result - only a single TPU.

However I now think the problem is the adapter board and I will explain why.... alongside trying to get the board working I've also been migrating my Frigate NVR installation to a new proxmox LXC container and I was constantly getting errors in the container like:

```
Traceback (most recent call last):
  File "examples/classify_image.py", line 121, in <module>
    main()
  File "examples/classify_image.py", line 71, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 163, in load_delegate
    library, str(e)))
ValueError: Failed to load delegate from libedgetpu.so.1
```

Googling a lot of people report this for USB adapters where they don't have it plugged in or have a dodgy USB cable (there isn't much info on this happening for PCIe devices)

In doing the above diagnostic I put the coral back into the old WiFI adapter I was using originally and suddenly the LXC container is no longer throwing the errors above and Frigate is working properly so it seems to suggest the dual adapter board is causing issues.

magic-blue-smoke, 2022-07-19:
@foxy82 
Could you please contact me using form at the bottom of page [here](https://mbs-shop.online/) for board replacement?

NickM-27, 2022-07-19:
Just to update, got mine in and both are showing up, thank you!! @magic-blue-smoke

ropeguru, 2022-07-27:
Mine is doing a cross country USA tour. JFK to LAX then it will have to come back to Virginia..

Hopefully it will get here soon so I can test..

magic-blue-smoke, 2022-07-27:
> Mine is doing a cross country USA tour. JFK to LAX then it will have to come back to Virginia..
> 
> Hopefully it will get here soon so I can test..

@ropeguru global logistics is painful sometimes and unfortunately this is something we can't control.

I'd recommend to double-check delivery address and contact DHL if there's a mistake or your package appears in FL or WA next time.

ropeguru, 2022-07-28:
> @ropeguru global logistics is painful sometimes and unfortunately this is something we can't control.
> 
> I'd recommend to double-check delivery address and contact DHL if there's a mistake or your package appears in FL or WA next time.

Oh, this was certainly not directed at you or any of the work you have done. This is my first time dealing with SF International shipping and there track is kind of useless on details.

EticWeb, 2022-10-22:
> From what I see here, the issue is with hardware. `lspci` should show two `System peripheral: Global Unichip Corp. Coral Edge TPU` lines, because TPUs are independent
>
> Could you please try [...]

Hi.
I face the same problem. I bought an adapter on https://www.makerfabs.com/dual-edge-tpu-adapter.html but since I have it I can succeed to see the two TPU.

EticWeb, 2022-10-25:
Hi. Could you confirm if the https://www.makerfabs.com/dual-edge-tpu-adapter.html  web store is the right place to buy your PCIe adapter for coral dual edge tpu?

magic-blue-smoke, 2022-10-26:
Hi @EticWeb 
Sorry for late reply, missed your first message.
Makerfabs is manufacturer and seller for my TPU adapter boards and Raspberry Pi CM4 TV Stick
Regarding issue you're having please contact me using form at the bottom of [home page](https://mbs-shop.online/)