magic-blue-smoke/Dual-Edge-TPU-Adapter #28: B+M Key Adapter mounting screw

cdrose, 2022-10-16:
I just received my B+M Key Adapters today from MakerFabs. They appear to work as expected under Windows with two Coral Devices showing up in Device Manager. My only concern is the screw that is shipped with the adapter for securing the M2 Coral does not have the right shape to secure them properly. The supplied screw has a round head which does not clamp down nicely on the M2 Coral and looks very close to not securing it at all. The correct fastener should have a flat mounting face on the underside.

magic-blue-smoke, 2022-10-17:
@cdrose Thanks for you feedback 
I'll review p/n for standoff and mounting screw and make changes for further batches

magic-blue-smoke, 2022-10-27:
After boards review decision was made to pause shipping and replace the screws

All new orders are being shipped with the new wider flat head screw  

[image]

akston, 2022-10-27:
Any suggestions for a solution (such as a part number from a reputable supplier) for those of us with the original design?

cdrose, 2022-10-30:
The new screw looks much better. As above, can you please let us know where we can order the correct part from.

magic-blue-smoke, 2022-10-30:
@akston @cdrose 
Thread is metric M2.5. Screw thread length is 4mm.

Aliexpress have those from different suppliers: [example 1](https://www.aliexpress.us/item/3256803373740847.html) [example 2](https://www.aliexpress.us/item/3256803148854258.html), make sure to pick the right thread size/length