magic-blue-smoke/Dual-Edge-TPU-Adapter #33: Have some option to connect Dual Edge TPU to laptop ?

antonbek89, 2023-02-24:
Hey
I just buy the Dual edge TPU have option to connect it to laptop ? with some adapter and magic blue smoke product ?

antonbek89, 2023-02-24:
a@debian:~$ lspci
00:00.0 Host bridge: Intel Corporation Haswell-ULT DRAM Controller (rev 0b)
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 0b)
00:04.0 Signal processing controller: Intel Corporation Haswell-ULT Thermal Subsystem (rev 0b)
00:14.0 USB controller: Intel Corporation 8 Series USB xHCI HC (rev 04)
00:16.0 Communication controller: Intel Corporation 8 Series HECI #0 (rev 04)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 1 (rev e4)
00:1c.2 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 3 (rev e4)
00:1c.3 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 4 (rev e4)
00:1c.4 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 5 (rev e4)
00:1d.0 USB controller: Intel Corporation 8 Series USB EHCI #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation 8 Series LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 8 Series SATA Controller 1 [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 8 Series SMBus Controller (rev 04)
00:1f.6 Signal processing controller: Intel Corporation 8 Series Thermal (rev 04)
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 10)
03:00.0 Network controller: Broadcom Limited BCM43142 802.11b/g/n (rev 01)
04:00.0 3D controller: NVIDIA Corporation GF117M [GeForce 610M/710M/810M/820M / GT 620M/625M/630M/720M] (rev a1)

magic-blue-smoke, 2023-02-24:
Hi @antonbek89 
If laptop has m.2 B- or M-key slot with NVMe interface (not SATA) it's theoretically possible to use Dual Edge TPU card.
However, I see some issues with this configuration:
- adapter with Dual Edge TPU card has high profile and most likely prevent case/NVMe lid from closing 
- even if there's enough clearance and laptop can be assembled back, there will be no space left for heatsink, which is highly recommended

Taking these issues into account I can't recommend or advise any of my adapters to be used with laptop.

Also, external USB-to-NVMe enclosures won't work with Coral TPU cards, as they only implement USB Mass Storage Class and corresponding PCIe funtionality.

antonbek89, 2023-02-24:
> Hi @antonbek89 If laptop has m.2 B- or M-key slot with NVMe interface (not SATA) it's theoretically possible to use Dual Edge TPU card. However, I see some issues with this configuration:
>
> * adapter [...]

Hi thanks for the answer 
So I guest I must to buy motherboard whoch support it

Did you can give me some example to motherboard which support all the requirements to install Dual TPU?

Not something fancy, just to understand what to search

nmajin, 2023-02-24:
> * even if there's enough clearance and laptop can be assembled back, there will be no space left for heatsink, which is highly recommended

Speaking of a heatsink, is there one you guys sell that fits this adapter/setup? Or would any heatsink for the coral work as long as their is clearance?