magic-blue-smoke/Dual-Edge-TPU-Adapter #35: PCIe adaptor for multiple mini-PCIe chips

dsnet, 2023-04-21:
Hi, thank you for your work!

This might be a short-sighted request, but reflects current production conditions.
At present, [the only module in stock](https://www.mouser.com/c/embedded-solutions/?q=edge%20tpu&product%20type=System-On-Modules%20-%20SOM&instock=y) is the single TPU in mini-PCIe form.

With a [passive adaptor like these](https://www.newegg.com/p/1W7-00R6-01DF8?Item=9SIAU62HXG4753), I was able to get them to connect to my server. However, this is highly wasteful occupying a multiple-lane PCIe slot with only a card that uses a single lane.

Would you be able to produce an adapter card that converts multiple PCIe lanes in a single slot to multiple mini-PCIe? Say, perhaps 4 mini-PCIe or any number that makes sense.

magic-blue-smoke, 2023-04-24:
Hi @dsnet 
Developing adapter for two Dual Edge TPU E-key cards is the first priority.
Having another options, like adapter for 4 B+M- or 4 A+E- or 4 miniPCIe is something I have in mind. Unsolved issue here is that today we have miniPCIe cards in stock, two weeks ago it was B+M- key and it's impossible to predict what's in stock next month.
There are two options I can see:
- Expensive: manufacture all three types of the cards
- Ugly: manufacture base adapter card with PCIe switch and board-to-board connector and a set of passive... adapters for adapter 

Also looking if there's a better idea

dsnet, 2023-04-24:
Another option: Sell PCIe cards with X number of Edge TPUs on it.

The buyer specifies X, you sell a card that matches X,
but don't guarantee whether:
* it's using [directly soldered modules](https://coral.ai/products/accelerator-module)
* pre-installed with single-core or double-core modules (via whatever connector), or
* whatever.

At the end of the day, the buyer just wants EdgeTPUs in PCIe format.
And so long as we can see all cores appear through `lspci`, then the buyer is happy.
You have a flexibility of choose how to make it work based on market conditions.

For this to work, I would expect the price to be somewhere within $20-$50 for every EdgeTPU.
I personally would be unwilling to pay more than double the cost of a single EdgeTPU module.

As an absurd pricing, consider the [Asus PCIe accelerator card](https://www.amazon.com/Accelerator-Based-Google-Inference-CRL-G18U-P3DF/dp/B08TZRX2WP), which costs a whopping $1310. It's made up internally of 8x dual-core modules. Each [dual core module costs $40](https://coral.ai/products/m2-accelerator-dual-edgetpu), so 8*40 should be $320. So for the adaptor, we're paying for a markup of $990. That's just not economical. For money is going to the adaptor than the AI modules themselves.

quantgeek, 2023-05-15:
I used the QNAP QM2-4P-384 Quad M.2 PCIe Card. It has a built-in PCIe bifurcation, so that it works on older motherboards. With the dual core modules and the m.2 B+M adapters, that gives me eight TPUs:

[image]

The QNAP is selling fairly cheap on ebay...

chino-lu, 2023-08-01:
> I used the QNAP QM2-4P-384 Quad M.2 PCIe Card. It has a built-in PCIe bifurcation, so that it works on older motherboards. With the dual core modules and the m.2 B+M adapters, that gives me eight TPUs:
> 
> [image]
> 
> The QNAP is selling fairly cheap on ebay. It comes with both the low-profile and high-profile PC brackets, so it installs nicely on a regular desktop...
> 
> https://www.ebay.com/itm/255953919797

could you let me know what adapter you are using exactly? Thanks.

quantgeek, 2023-08-01:
I bought four of the following adapters:

 

https://www.makerfabs.com/dual-edge-tpu-adapter-m2-2280-b-m-key.html

 

Regards,

 

QG

 

 <https://user-images.githubusercontent.com/45181914/238467056-a07a366e-00fc-4c2b-9ad4-30c99ac9d6d7.png> 

https://www.ebay.com/itm/255953919797