magic-blue-smoke/Dual-Edge-TPU-Adapter #37: Passthrough adapter to ESXi VM not working

sjkjs, 2023-05-19:
I'm running ESXi 8.0.1 on an HP DL380p Gen8 and want to pass through the Coral devices to a VM.

If I SSH into the ESXi server and run lspci, it shows:

`0000:0c:00.0 Non-VGA unclassified device: Global Unichip Corp. Coral Edge TPU
0000:0d:00.0 Non-VGA unclassified device: Global Unichip Corp. Coral Edge TPU`

Under ESXi PCI devices I see that passthrough is not supported:

[image]

Through vCenter it says "This device cannot be made available for VMs to use" when selecting it.

[image]

If I click toggle passthrough, it displays an error saying "This device cannot be made available for VMs to use"

Has anyone gotten this to work before?

jahegu, 2023-06-26:
same problem, any new?

magic-blue-smoke, 2023-07-01:
Passthrough requires VM-d/iommu support and VM-d/iommu enabled in BIOS. Is it something what was already tried?

sjkjs, 2023-07-01:
Yes, VT-d was enabled and other devices are successfully being used in passthrough mode.

I've since given up and sold the Coral so I can't test any further.