magic-blue-smoke/Dual-Edge-TPU-Adapter #40: Prevents MB Power Up

anernest, 2023-07-12:
I just received 2 of these units. With the Dual Edge TPU inserted, and the PCIe card inserted into the motherboard (ASUS Prime B540M A-AX with Ryzen 7950x CPU), will not power on.

magic-blue-smoke, 2023-07-12:
@anernest sorry to hear about the issue you have. While these adapters are using standard PCIe switch, there might be incompatibilities with particular systems.

Could you please check following things:
- if using PCIe x16 slot: adapter inserted correctly, not flipped 180 degrees (unlikely, but sometimes happens)
- if using PCIe x16 slot: check BIOS settings, if there are options for slot to fall back to PCIe 2.0 and x1 lane

If nothing helps, we'll declare this configuration as non-compatible and proceed with adapters return

anernest, 2023-07-21:
@magic-blue-smoke , just call me "unlikely" rather than any other words that might spring up more readily - I had them flipped 180.

anernest, 2023-07-21:
Operator error.

closed 2023-07-21 by anernest