magic-blue-smoke/Dual-Edge-TPU-Adapter #41: Optiplex 3000 Micro SSD+Wifi+EdgeTPU 2 M2 slots

0x3333, 2023-08-13:
Hi! I have a Dell Optiplex 300 Micro, with 2 M2 slots, one for SSD and another for Wifi+Bluetooth card.

I'd like to install an Edge TPU or Dual Edge TPU on this machine.

Do you have an adapter for this scenario?

I have only these 2 slots available:

[image]

yanfali, 2023-08-13:
I'm using an optiplex micro using the m.2 b+m adapter for a dual coral PCI
tpu. It's been working well for frigate.

0x3333, 2023-08-14:
> I'm using an optiplex micro using the m.2 b+m adapter for a dual coral PCI tpu. It's been working well for frigate.
> […](#)

Cool, you are using in the SSD slot, right?

yanfali, 2023-08-14:
Correct.
```
04:00.0 System peripheral: Global Unichip Corp. Coral Edge TPU (prog-if ff)
        Subsystem: Global Unichip Corp. Coral Edge TPU
        Flags: bus master, fast devsel, latency 0, IRQ 19
        Memory at f0300000 (64-bit, prefetchable) [size=16K]
        Memory at f0200000 (64-bit, prefetchable) [size=1M]
        Capabilities: [80] Express Endpoint, MSI 00
        Capabilities: [d0] MSI-X: Enable+ Count=128 Masked-
        Capabilities: [e0] MSI: Enable- Count=1/32 Maskable- 64bit+
        Capabilities: [f8] Power Management version 3
        Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
        Capabilities: [108] Latency Tolerance Reporting
        Capabilities: [110] L1 PM Substates
        Capabilities: [200] Advanced Error Reporting
        Kernel driver in use: apex
        Kernel modules: apex

05:00.0 System peripheral: Global Unichip Corp. Coral Edge TPU (prog-if ff)
        Subsystem: Global Unichip Corp. Coral Edge TPU
        Flags: bus master, fast devsel, latency 0, IRQ 19
        Memory at f0100000 (64-bit, prefetchable) [size=16K]
        Memory at f0000000 (64-bit, prefetchable) [size=1M]
        Capabilities: [80] Express Endpoint, MSI 00
        Capabilities: [d0] MSI-X: Enable+ Count=128 Masked-
        Capabilities: [e0] MSI: Enable- Count=1/32 Maskable- 64bit+
        Capabilities: [f8] Power Management version 3
        Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
        Capabilities: [108] Latency Tolerance Reporting
        Capabilities: [110] L1 PM Substates
        Capabilities: [200] Advanced Error Reporting
        Kernel driver in use: apex
        Kernel modules: apex
```