magic-blue-smoke/Dual-Edge-TPU-Adapter #47: Dual Edge TPU Adapter for multiple TPU cards

n3de, 2023-12-08:
Hey,
is there any update for the Dual Edge TPU Adapter for multiple TPU cards? When will it be availible?
best regards

magic-blue-smoke, 2024-01-11:
Hi @n3de 
Particular PCIe switch IC I used for prototype doesn't work well. New best estimate is around April

n3de, 2024-01-13:
Thanks for the update. Hopefully the new version will work well :-)

magic-blue-smoke, 2024-06-06:
Hi @n3de 
Prototype, redesigned with another PCIe switch IC, was manufactured and shipped to me. Hopefully I can share some news next month

jaburges, 2024-06-17:
Yey! So excited about this! Got 2 x Dual TPUs ready to test.
Since we first chatted Unraid has brought out a native coral driver package.

All systems go!

G2G2G2G, 2024-10-04:
Looking forward to this 2030

n3de, 2024-10-06:
> Hi @n3de Prototype, redesigned with another PCIe switch IC, was manufactured and shipped to me. Hopefully I can share some news next month

Any updates?

magic-blue-smoke, 2024-10-07:
@n3de it was tricky to make it work but I was able to successfully run [multiple TPU test](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/21#issuecomment-2356929753)
I plan to manufacture a small batch of these boards this month and make a "soft launch". This means buyer understands that adapter was tested on a limited number of systems and I realize there might be higher return rate

n3de, 2024-10-08:
Great to hear. Will you inform through the waiting list, or how will the soft launch be? If I get a board and I will test it on my system. Do you use a specific test and is it documentated?