magic-blue-smoke/Dual-Edge-TPU-Adapter #50: Dual Edge TPU Adapter Causing PCI issues and BSOD Server 2022

Bachperson, 2023-12-27:
Hello, I recently got a Dual TPU adapter and i plugged it in, installed the drivers and it worked great in my Lenovo SR250 blade server. I am using it for AI detection in Blue Iris. Both TPU's were detected and I checked both TPU temps and everything seemed good (in the 30-40c range). The server ran for about 5 min and than had a BSOD. I checked the seating of both the TPU and the pcie slot and gave everything a good blow of air to make sure there were no dust or contaminants. Loaded the server back up but after some time it did the same thing. Here is a log dump from what my server says is erroring out. Any help would be appreciated because when it works, it works amazingly.

0		Power	FQXSPPW0008I	Host Power has been turned off.	December 27, 2023 2:18:13 PM
1		System	FQXSPPW0009I	Host Power has been Power Cycled.	December 27, 2023 2:17:38 PM
2		System	FQXSPIO0011N	An Uncorrectable Error has occurred on CPUs.	December 27, 2023 2:17:05 PM
3		Memory	FQXSFMA0006I	Unqualified DIMM 1 has been detected, the DIMM serial number is 1AA6ECDC-V20.	December 27, 2023 2:04:08 PM
4		Memory	FQXSFMA0006I	Unqualified DIMM 2 has been detected, the DIMM serial number is 1AADD8FF-V20.	December 27, 2023 2:04:08 PM
5		Memory	FQXSFMA0006I	Unqualified DIMM 3 has been detected, the DIMM serial number is 1AAD66FB-V20.	December 27, 2023 2:03:58 PM
6		Memory	FQXSFMA0006I	Unqualified DIMM 4 has been detected, the DIMM serial number is 1AA6EBA8-V20.	December 27, 2023 2:03:58 PM
11		Disks	FQXSPSD0000I	The M2 Drive has been added.	December 27, 2023 2:02:48 PM
12		Power	FQXSPPW0008I	Host Power has been turned off.	December 27, 2023 1:26:26 PM
13		System	FQXSPPW0009I	Host Power has been Power Cycled.	December 27, 2023 2:11:11 AM
14		System	FQXSPPW0009I	Host Power has been Power Cycled.	December 27, 2023 2:05:32 AM
15		System	FQXSPIO2006I	System ThinkSystem SR250 has recovered from an NMI.	December 27, 2023 2:01:17 AM
16		System	FQXSPPW0009I	Host Power has been Power Cycled.	December 27, 2023 2:00:51 AM
17		System	FQXSPPW0009I	Host Power has been Power Cycled.	December 27, 2023 2:00:36 AM
18		System	FQXSPIO0006N	A software NMI has occurred on system ThinkSystem SR250.	December 27, 2023 1:58:59 AM
19		System	FQXSPIO0015M	Fault in slot 2 on system ThinkSystem SR250.	December 27, 2023 1:58:56 AM
20		System	FQXSFIO0010M	An Uncorrectable PCIe Error has Occurred at Bus 0000 Device 01 Function 00. The Vendor ID for the device is 8086 and the Device ID is 1901. The Physical slot number is 2.	December 27, 2023 1:58:53 AM
21		System	FQXSFIO0010M	An Uncorrectable PCIe Error has Occurred at Bus 0000 Device 01 Function 00. The Vendor ID for the device is 8086 and the Device ID is 1901. The Physical slot number is 2.	December 27, 2023 1:58:52 AM
22		System	FQXSPIO2015I	Fault condition removed on slot 2 on system ThinkSystem SR250.	December 27, 2023 1:58:51 AM
23		System	FQXSPPW0009I	Host Power has been Power Cycled.	December 27, 2023 12:39:19 AM
24		System	FQXSPIO2006I	System ThinkSystem SR250 has recovered from an NMI.	December 27, 2023 12:19:37 AM
25		System	FQXSPPW0009I	Host Power has been Power Cycled.	December 27, 2023 12:19:09 AM
26		System	FQXSPPW0009I	Host Power has been Power Cycled.	December 27, 2023 12:18:53 AM
27		System	FQXSPIO0006N	A software NMI has occurred on system ThinkSystem SR250.	December 27, 2023 12:17:17 AM
28		System	FQXSPIO0015M	Fault in slot 2 on system ThinkSystem SR250.	December 27, 2023 12:17:17 AM
29		System	FQXSFIO0010M	An Uncorrectable PCIe Error has Occurred at Bus 0000 Device 01 Function 00. The Vendor ID for the device is 8086 and the Device ID is 1901. The Physical slot number is 2.	December 27, 2023 12:17:14 AM
30		System	FQXSPCA2015I	Sensor CPU Overtemp has deasserted the transition from normal to non-critical state.	December 27, 2023 12:07:48 AM
31		System	FQXSPCA0015J	Sensor CPU Overtemp has transitioned from normal to non-critical state.	December 27, 2023 12:07:45 AM
32		System	FQXSPPW0009I	Host Power has been Power Cycled.	December 27, 2023 12:06:07 AM
33		System	FQXSPIO0011N	An Uncorrectable Error has occurred on CPUs.	December 27, 2023 12:05

Bachperson, 2023-12-28:
Update, Seems like when there is no load on the TPU it does not cause a crash on the system its only when there is load on it it causes BSOD.

Bachperson, 2023-12-30:
Update, tried everything in different computer and it runs stable with all the same software. Tested all various settings in blade server to no avail. Extremely frustrating as it works for like 30 min and works great but will inevitably BSOD. I have tried it in both pcie slots in my Lenovo SR250, forcing gen 2 on the pcie buss enabling pcie error recovery etc. any help or ideas would be greatly appreciated.

markfrancisonly, 2023-12-30:
Try experimenting with the PCIe and adapter card slot settings in your BIOS. I've experience a problem with a new motherboard where inserting an older generation card into a slot set as Auto / PCIe 5.0 caused the system to fail. Manually configuring the slot to operate at a lower generation allowed the system to boot without any problems. 

My problem manifest on a low quality asus motherboard with an uncommon w680 chipset. I ultimately got rid of it

markmghali, 2024-01-23:
I had the same issuers on an unraid server. Any load would cause my whole server to stop working. It was very frustrating. I ended up returning it. He said he was going to make another version but haven
t heard about it yet

Bachperson, 2024-01-23:
I pretty much ended up doin the same exact thing, I got the mini pcie tpu with this pcie to mini pcie adapter from Amazon (https://www.amazon.com/dp/B07JBCL1CJ?ref=ppx_pop_mob_ap_share) and haven't had any problems, works totally fine, so who knows.

markmghali, 2024-01-23:
> I pretty much ended up doin the same exact thing, I got the mini pcie tpu with this pcie to mini pcie adapter from Amazon (https://www.amazon.com/dp/B07JBCL1CJ?ref=ppx_pop_mob_ap_share) and haven't had any problems, works totally fine, so who knows. 

Yeah! That's the same adaptor I am using. You only get one tpu correct?

mateuszdrab, 2024-05-01:
Same issue here, just put a dual TPU adapter in the M2 adapter card which sits in a M2 NVMe PCIe adapter from starcom housed inside a dl380 Gen9.

Running Server 2022, I get a BSOD when I try to passthrough the device into the VM (starting the VM, mounting the device back to the host) and also when installing drivers for the chipset/rescanning devices.

Is the only solution to send the card back and instead get a single TPU mini PCIe adapter?

Bachperson, 2024-05-01:
Yep unfortunately I ended up getting the single tpu and pci adapter (https://www.amazon.com/dp/B07JBCL1CJ?ref=ppx_pop_mob_ap_share) and it's still goin no problems. I saw there was an update on codeproject Ai that added dual tpu support but I doubt that will help with this issue at all as it seems to be more hardware related than software but who knows 🤷.

mateuszdrab, 2024-05-01:
> Yep unfortunately I ended up getting the single tpu and pci adapter (https://www.amazon.com/dp/B07JBCL1CJ?ref=ppx_pop_mob_ap_share) and it's still goin no problems. I saw there was an update on [...]

Well, I'm trying to understand if the issue is the dual TPU accelerator or the adapter card.

I can return the accelerator easily, not sure about the adapter card.

markmghali, 2024-05-01:
My issue was the adapter card specifically. My TPU was not the issue, to my
knowledge

Bachperson, 2024-05-01:
Unfortunately I'm pretty sure it's the adapter card or more likely an incompatibility between the adapter card and the PC because I tried it in another computer and it worked fine no problem.

mateuszdrab, 2024-05-01:
> Unfortunately I'm pretty sure it's the adapter card or more likely an incompatibility between the adapter card and the PC because I tried it in another computer and it worked fine no problem. 

That's interesting, so you've managed to get both TPUs to work in another PC through the adapter?

What were the specs of that PC? I wonder if it's something chipset/PCIe related since the card does does splitting of one PCIe lane.

mateuszdrab, 2024-05-03:
> @mateuszdrab unfortunately, in rare cases, there are compatibility issues with some motherboards.
> To see if this is a compatibility issue or DoA adapter, could you please test it with another, preferably desktop PC?

I've tested it on two HP gen 9 servers. 
I'll try to test it on one of my desktops but it's even older as my new one only has one slot which is taken by the GPU.

Bachperson, 2024-05-03:
Sorry for the late reply, the mobo I used that it did work on is a MSI MPG Z690 CARBON Wifi (MS-7D30)

mateuszdrab, 2024-05-03:
> Sorry for the late reply, the mobo I used that it did work on is a MSI MPG Z690 CARBON Wifi (MS-7D30)

Thanks

Are you using them on the host OS or passing through into a VM? I wonder if you could do a hyper-v passthrough test (if you're running Windows)

magic-blue-smoke, 2024-05-05:
@mateuszdrab HP servers have demonstrated higher incompatibility cases with only a few series having no issues. For this reason it's better to test with desktop PC.

I was collecting incompatible hardware cases, need to sum it up and publish

mateuszdrab, 2024-05-05:
> @mateuszdrab HP servers have demonstrated higher incompatibility cases with only a few series having no issues. For this reason it's better to test with desktop PC.
> 
> I was collecting incompatible hardware cases, need to sum it up and publish

I've been trying to get it working on my desktop PC. 
Haven't had a BSOD yet, but not tested hyper-v passthrough as some uefi setting is preventing it from being assignable to a vm.

mateuszdrab, 2024-05-06:
After further testing, running some inferencing and upgrading Windows on the PC, I've had no BSODs so far.

Like like the setting needed to enable DDA in hyper-v is missing so I was not able to try it. 

Perhaps the same feature is what causes the BSOD.