magic-blue-smoke/Dual-Edge-TPU-Adapter #51: Can I use the PCIe adapter w/QNAP

diegoroccia, 2023-12-28:
Hi! 
I have a [QNAP TS-264](https://www.qnap.com/en/product/ts-264) which is already working well with a Coral USB external TPU. Now I would like to find a way to use it with the M.2 Accelerator with Dual Edge TPU (Key E) which I have sitting in the drawer. 

Since the QNAP has one PCIe Gen 3 x2 expansion slot, I was wondering if I can use the [Dual Edge TPU Adapter - PCIe x1 Low Profile](https://www.makerfabs.com/dual-edge-tpu-adapter.html) and if this will allow me to see both TPUs from the QNAP

magic-blue-smoke, 2024-01-11:
Hi @diegoroccia 
QNAP TS-264 is compatible with m.2 M-key Coral TPU card according to this [Official Compatibility List](https://www.qnap.com/en-us/compatibility/?device_category=ai%20accelerator&brand=coral)
What is under question is if adapter can:
- fit mechanically (increased height compared to m.2 drive)
- QNAP software can use more that one TPU

If you have some enthusiasm to test this config, send me a message using a form at the bottom of [this page](https://mbs-shop.online/)

diegoroccia, 2024-01-14:
Hey @magic-blue-smoke thank you for your answer. Since 
- I already have the M2 E-Key dual TPU
- I am using the M2 slots for SSDs already

I was hoping to be able to make use of the PCIe slot and get 2 TPUs to work.

Are you saying that you expect this to not work?