magic-blue-smoke/Dual-Edge-TPU-Adapter #52: coral m.2 dual speed question

eflydata, 2024-03-19:
I purchased m.2 BM version, But one of them is slow.

[image]

`lspci

03:00.0 PCI bridge: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch
04:03.0 PCI bridge: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch
04:07.0 PCI bridge: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch
05:00.0 System peripheral: Global Unichip Corp. Coral Edge TPU
06:00.0 System peripheral: Global Unichip Corp. Coral Edge TPU
`
`sudo lspci -n -d 1ac1:089a -vvv | grep -i width

LnkCap: Port #1, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <64ns, L1 <1us
LnkSta: Speed 5GT/s, Width x1
LnkCap: Port #1, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <64ns, L1 <1us
LnkSta: Speed 2.5GT/s (downgraded), Width x1
`

pocki, 2024-03-19:
Are you using the same model on both TPUs?

eflydata, 2024-03-19:
> Are you using the same model on both TPUs?

Yes,I Use Coral M.2 Dual.

pocki, 2024-03-19:
Sorry, I mean in frigate. Have you assigned both TPUs the same detector/interference model?

magic-blue-smoke, 2024-03-20:
@eflydata link speed downgrade is not something what's expected here. Please try re-inserting TPU card and, if possible, try with another TPU card. If it won't help, please contact using form at very bottom of [this page](https://mbs-shop.online/)