magic-blue-smoke/Dual-Edge-TPU-Adapter #53: Coral M.2 Accelerator dual edge tpu with Dual Edge TPU Adapter - PCIe x1 Low Profile only one tpu working

duindain, 2024-03-27:
Hi,

I've bought a m.2 dual edge accelerator and an adapter from makerfab

I've just got all the cameras working and running frigate and I'm getting constant reboots of frigate saying it can't find one of the tpus

I'm running frigate in a docker container

dmesg looks like its reporting an error from the adapter/accelerator possibly?

```
ls -l /dev/apex*
crw-rw---- 1 root apex 120, 0 Mar 25 18:23 /dev/apex_0
crw-rw---- 1 root apex 120, 1 Mar 25 18:23 /dev/apex_1
```

```
ls /sys/class/apex/
apex_0  apex_1
```

```
dmesg | grep apex
[   35.356036] apex 0000:05:00.0: enabling device (0000 -> 0002)
[   35.371387] apex 0000:06:00.0: enabling device (0000 -> 0002)
[   40.512237] apex 0000:05:00.0: Apex performance not throttled due to temperature
[   48.172225] apex 0000:06:00.0: RAM did not enable within timeout (12000 ms)
[   53.312236] apex 0000:06:00.0: Apex performance not throttled due to temperature
[   58.432237] apex 0000:06:00.0: Apex performance not throttled due to temperature
[   63.552236] apex 0000:06:00.0: Apex performance not throttled due to temperature
[   68.672243] apex 0000:06:00.0: Apex performance not throttled due to temperature
[   83.300290] apex 0000:06:00.0: RAM did not enable within timeout (12000 ms)
[   83.300301] apex 0000:06:00.0: Error in device open cb: -110
[   83.300315] apex 0000:06:00.0: Apex performance not throttled due to temperature
[   88.384283] apex 0000:06:00.0: Apex performance not throttled due to temperature
[   93.504246] apex 0000:06:00.0: Apex performance not throttled due to temperature
[   98.628059] apex 0000:06:00.0: Apex performance not throttled due to temperature
[  115.671423] apex 0000:06:00.0: RAM did not enable within timeout (12000 ms)
[  115.671430] apex 0000:06:00.0: Error in device open cb: -110
[  115.671442] apex 0000:06:00.0: Apex performance not throttled due to temperature
[  120.895350] apex 0000:06:00.0: Apex performance not throttled due to temperature
[  126.015273] apex 0000:06:00.0: Apex performance not throttled due to temperature
[  131.135234] apex 0000:06:00.0: Apex performance not throttled due to temperature
```

frigate docker compose file
```
version: "3.9"
services:
  frigate:
    container_name: frigate
    privileged: true # this may not be necessary for all setups
    restart: unless-stopped
    image: ghcr.io/blakeblackshear/frigate:stable
    shm_size: "850mb" # update for your cameras based on calculation above
    devices:
      #- /dev/bus/usb:/dev/bus/usb # passes the USB Coral, needs to be modified for other versions
      - /dev/apex_0:/dev/apex_0 # passes a PCIe Coral, follow driver instructions here https://coral.ai/docs/m2/get-started/#2a-on-linux
      - /dev/apex_1:/dev/apex_1
      #- /dev/dri/renderD128 # for intel hwaccel, needs to be updated for your hardware
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /home/user/Software/Scripts/docker/frigate/config/frigate.yml:/config/config.yml
      - /home/user/Software/Scripts/docker/frigate/config/go2rtc:/config/go2rtc
      - /mnt/CamFootage:/media/frigate
      - /home/user/Software/Scripts/docker/frigate:/db
      - type: tmpfs # Optional: 1GB of memory, reduces SSD/SD Card wear
        target: /tmp/cache
        tmpfs:
          size: 1000000000
[... 88 lines cut ...]
2024-03-26 01:50:23.252760402  [INFO] Service NGINX exited with code 0 (by signal 0)
s6-rc: info: service nginx successfully stopped
s6-rc: info: service nginx-log: stopping
s6-rc: info: service frigate: stopping
2024-03-26 01:50:23.258859999  [2024-03-26 01:50:23] frigate.app                    INFO    : Stopping...
s6-rc: info: service nginx-log successfully stopped
2024-03-26 01:50:23.259005072  [2024-03-26 01:50:23] root                           INFO    : Waiting for detection process to exit gracefully...
2024-03-26 01:50:23.259067950  [2024-03-26 01:50:23] frigate.stats                  INFO    : Exiting stats emitter...
2024-03-26 01:50:23.259199507  [2024-03-26 01:50:23] frigate.watchdog               INFO    : Exiting watchdog...
2024-03-26 01:50:23.259248820  [2024-03-26 01:50:23] frigate.ptz.autotrack          INFO    : Exiting autotracker...
2024-03-26 01:50:23.259361050  [2024-03-26 01:50:23] frigate.storage                INFO    : Exiting storage maintainer...
2024-03-26 01:50:23.259403901  [2024-03-26 01:50:23] frigate.record.cleanup         INFO    : Exiting recording cleanup...
2024-03-26 01:50:23.259478781  [2024-03-26 01:50:23] frigate.events.cleanup         INFO    : Exiting event cleanup...
2024-03-26 01:50:23.260938894  [2024-03-26 01:50:23] detector.coral1                INFO    : Signal to exit detection process...
2024-03-26 01:50:23.263547072  Fatal Python error: Segmentation fault
```

If i comment out in the frigate config `- /dev/apex_1:/dev/apex_1` and restart frigate container it runs and stops rebooting and dmesg stops reporting 
`[  115.671430] apex 0000:06:00.0: Error in device open cb: -110`

I've removed the adapter and checked its seated well and no dust and reinserted it to the pci port

CPU: Ryzen 7 5700G
Motherboard:  B550M Steel Legend
GPU: Onboard

duindain, 2024-04-07:
In docker passing through any of these works fine individually when the frigate config is only using pcie:0
```
- /dev/apex_0:/dev/apex_1
- /dev/apex_1:/dev/apex_0
- /dev/apex_1:/dev/apex_1
- /dev/apex_0:/dev/apex_0
```

If i set the frigate config to use pcie:1 it fails

I don't have a heatsink atm, i can add one

magic-blue-smoke, 2024-04-09:
@duindain please try it with heatsink, as it's needed anyways. If it won't help, we'll consider adapter replacement

duindain, 2024-04-12:
I've put a passive heat sink on with some thermal joining pad, its definitely not high quality but the case is well ventilated, has a 120mm fan and its fairly cool here atm 14-20c ambient

I'm not sure if this is accurate or how you are meant to check (There didnt seem to be much info out there) but i get this values

When passing through just apex_0 from docker
cat /sys/class/apex/apex_0/temp 48300 in a range so 46-48 degrees c
cat /sys/class/apex/apex_1/temp -89700 this seems to always return this number

I assume the -89700 is because its not being used? or from just not running

I've tried a few combinations but apex_1 always seems to return that -89700 regardless

magic-blue-smoke, 2024-04-21:
@duindain feels like something's wrong with either TPU card or adapter itself. If you can't inspect flipchips on your TPU card with microscope or try another card, we can try to replace adapter

duindain, 2024-04-29:
@magic-blue-smoke unfortunately the best i have is a magnifying lens and i cant see anything looking broken or badly soldered, I don't have another card to try

magic-blue-smoke, 2024-05-02:
@duindain we can try adapter board replacement. Could you contact me using a [contact form](https://mbs-shop.online/) at the bottom of the page?

duindain, 2024-05-03:
ty, i"ve sent a message with order details and other info

duindain, 2024-06-13:
I've received the new adapter unfortunately the coral is behaving the same as before with one temperature sensor reporting an out of bounds value -89700
If i enable both tpus frigate continually crashes as before but i can enable one fine

I've send an RMA request for the coral
Is there anything else to try at this point?

duindain, 2024-06-18:
Mouser rejected the warranty request

markmghali, 2024-08-12:
curious my TPU is also showing a negative temp value and crashing my whole system 
 Coral Temp2: -89.70C