magic-blue-smoke/Dual-Edge-TPU-Adapter #56: Compatibility question with SBCs

skol101, 2024-04-03:
Hello,

I'm curious whether
1. m.2 A+E version will allow to use both TPUs on RPI 5 via Pineberry PI HAT e key https://pineberrypi.com/products/hat-ai-for-raspberry-pi-5 ?

2. bm version will allow to use both TPUs on Khadas VIM3 https://docs.khadas.com/products/sbc/vim3/hardware/interfaces ?

skol101, 2024-04-04:
@magic-blue-smoke would be interested in hearing your comments.

magic-blue-smoke, 2024-04-04:
Hi @skol101 

Thanks for your interest in adapters
Regarding your questions:
1a. I don't have A+E key adapter in production at the moment, so E-key hat is not an option. 

1b. B+M key adapter with "HatDrive! Top" might work, however it's not tested and the main concern here is power. Two TPUs with PCIe switch can exceed 3A current capability of HatDrive board

 2. Same as (1b): power might be an issue, as this is a tiny SBC

skol101, 2024-04-05:
Ok, we'll test google's https://coral.ai/products/m2-accelerator-dual-edgetpu/#tech-specs E.key with pineberry's hat

closed 2024-04-05 by skol101