magic-blue-smoke/Dual-Edge-TPU-Adapter #58: Low profile PCIe x1 version doesn't work on Proxmox v8.2

jackj120, 2024-05-04:
Hi i've tried to install the needed drivers following this guide: https://forum.proxmox.com/threads/coral-tpu-pcie-on-proxmox.140128/ , https://forum.proxmox.com/threads/coral-tpu-pcie-on-proxmox.140128/  and the coral websites   but nothing... 
When i type 
lspci -nn | grep 089a nothing is showed . I saw that sometimes i have problems with dkms . 
Also with no guides software installed when i run lspci i don't see nothing , i think that something must be showed . 
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD/ATI] RD9x0/RX980 Host Bridge (rev 02)
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD/ATI] RD890S/RD990 I/O Memory Management Unit (IOMMU)
00:02.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GFX port 0)
00:04.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 0)
00:09.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 4)
00:11.0 SATA controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 SATA Controller [AHCI mode] (rev 40)
00:12.0 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller
00:12.2 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB EHCI Controller
00:13.0 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller
00:13.2 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB EHCI Controller
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 SMBus Controller (rev 42)
00:14.1 IDE interface: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 IDE Controller (rev 40)
00:14.2 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 Azalia (Intel HDA) (rev 40)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 LPC host controller (rev 40)
00:14.4 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 PCI to PCI Bridge (rev 40)
00:14.5 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI2 Controller
00:16.0 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller
00:16.2 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB EHCI Controller
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h Processor Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h Processor Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h Processor Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h Processor Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h Processor Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h Processor Function 5
01:00.0 VGA compatible controller: NVIDIA Corporation GM204 [GeForce GTX 970] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GM204 High Definition Audio Controller (rev a1)
02:00.0 USB controller: VIA Technologies, Inc. VL805/806 xHCI USB 3.0 Controller (rev 01)
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)
Idk what it is hope in some help.

magic-blue-smoke, 2024-05-05:
Hi @jackj120 
- is this lspci for host or vm?
- did you set up PCIe passthrough for TPUs?

jackj120, 2024-05-05:
Its lspci from the host(then i've to do this for my frigate vm) . I saw that i've problem with gasket-dkms idk if it's related : 
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/gasket/1.0/source/dkms.conf)
Error! Bad return status for module build on kernel: 6.5.13-5-pve (x86_64)
Consult /var/lib/dkms/gasket/1.0/build/make.log for more information.
dpkg: error processing package gasket-dkms (--configure):
 installed gasket-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 gasket-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

with nano /var/lib/dkms/gasket/1.0/build/make.log return this 
DKMS make.log for gasket-1.0 for kernel 6.5.13-5-pve (x86_64)
Sun May  5 08:54:29 PM CEST 2024
make: Entering directory '/usr/src/linux-headers-6.5.13-5-pve'
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_core.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_ioctl.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_interrupt.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_page_table.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_sysfs.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/apex_driver.o
/var/lib/dkms/gasket/1.0/build/gasket_core.c: In function ‘gasket_register_device’:
/var/lib/dkms/gasket/1.0/build/gasket_core.c:1841:41: error: passing argument 1 of ‘class_create’ from incompatible p>
 1841 |                 class_create(driver_desc->module, driver_desc->name);
      |                              ~~~~~~~~~~~^~~~~~~~
      |                                         |
      |                                         struct module *
In file included from ./include/linux/device.h:31,
                 from ./include/linux/cdev.h:8,
                 from /var/lib/dkms/gasket/1.0/build/gasket_core.h:11,
                 from /var/lib/dkms/gasket/1.0/build/gasket_core.c:12:
./include/linux/device/class.h:230:54: note: expected ‘const char *’ but argument is of type ‘struct module *’
  230 | struct class * __must_check class_create(const char *name);
 ^~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/gasket/1.0/build/gasket_core.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.5.13-5-pve/Makefile:2039: /var/lib/dkms/gasket/1.0/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.5.13-5-pve'

Calimerorulez, 2024-05-05:
@jackj120 

https://github.com/google-coral/edgetpu/issues/695

Maybe?

jackj120, 2024-05-05:
Yeah i think it's the same problem, i saw it... So i need only to wait the fix?

Calimerorulez, 2024-05-05:
@jackj120 

I think a solution is that you build the driver by yourself, but that is fairly complicated.

This is a comment about which commands to use:

https://github.com/google-coral/edgetpu/issues/808#issuecomment-1909019568

jackj120, 2024-05-05:
@Calimerorulez 
I'll try this comment . Thank for the help .

jackj120, 2024-05-05:
@Calimerorulez 
Ok i've followed the instruction and the installation worked fine but i don't see the coral with lspci

i've tried also this lspci -nn | grep 089a but nothing it's returned

lspci -v
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD/ATI] RD9x0/RX980 Host Bridge (rev 02)
        Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] RD9x0/RX980 Host Bridge
        Flags: fast devsel, NUMA node 0, IOMMU group 0
        Capabilities: [f0] HyperTransport: MSI Mapping Enable+ Fixed+
        Capabilities: [c4] HyperTransport: Slave or Primary Interface
        Capabilities: [40] HyperTransport: Retry Mode
        Capabilities: [54] HyperTransport: UnitID Clumping
        Capabilities: [9c] HyperTransport: #1a
        Capabilities: [70] MSI: Enable- Count=1/4 Maskable- 64bit-

00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD/ATI] RD890S/RD990 I/O Memory Management Unit (IOMMU)
        Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] RD890S/RD990 I/O Memory Management Unit (IOMMU)
        Flags: fast devsel, IRQ 26, NUMA node 0
        Capabilities: [40] Secure device <?>
        Capabilities: [54] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [64] HyperTransport: MSI Mapping Enable+ Fixed+

00:02.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GFX port 0) (prog-if 00 [Normal decode])
        Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GFX port 0)
        Flags: bus master, fast devsel, latency 0, IRQ 27, NUMA node 0, IOMMU group 1
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        I/O behind bridge: e000-efff [size=4K] [16-bit]
        Memory behind bridge: fd000000-fe0fffff [size=17M] [32-bit]
        Prefetchable memory behind bridge: c0000000-d1ffffff [size=288M] [32-bit]
        Capabilities: [50] Power Management version 3
        Capabilities: [58] Express Root Port (Slot+), MSI 00
        Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit-
        Capabilities: [b0] Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GFX port 0)
        Capabilities: [b8] HyperTransport: MSI Mapping Enable+ Fixed+
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [190] Access Control Services
        Kernel driver in use: pcieport

00:04.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 0) (prog-if 00 [Normal decode])
        Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 0)
        Flags: bus master, fast devsel, latency 0, IRQ 28, NUMA node 0, IOMMU group 2
        Bus: primary=00, secondary=02, subordinate=02, sec-latency=0
        I/O behind bridge: [disabled] [32-bit]
        Memory behind bridge: fe200000-fe2fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: [disabled] [64-bit]
        Capabilities: [50] Power Management version 3
        Capabilities: [58] Express Root Port (Slot+), MSI 00
        Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit-
        Capabilities: [b0] Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 0)
        Capabilities: [b8] HyperTransport: MSI Mapping Enable+ Fixed+
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [190] Access Control Services
        Kernel driver in use: pcieport

00:09.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 4) (prog-if 00 [Normal decode])
        Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 4)
        Flags: bus master, fast devsel, latency 0, IRQ 30, NUMA node 0, IOMMU group 3
        Bus: primary=00, secondary=03, subordinate=03, sec-latency=0
        I/O behind bridge: d000-dfff [size=4K] [16-bit]
        Memory behind bridge: fe100000-fe1fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: d2100000-d21fffff [size=1M] [32-bit]
        Capabilities: [50] Power Management version 3
        Capabilities: [58] Express Root Port (Slot+), MSI 00
        Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit-
        Capabilities: [b0] Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 4)
        Capabilities: [b8] HyperTransport: MSI Mapping Enable+ Fixed+
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [190] Access Control Services
        Kernel driver in use: pcieport
[... 160 lines cut ...]
        Flags: fast devsel, IRQ 33, NUMA node 0, IOMMU group 15
        Memory at fe200000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: [80] Power Management version 3
        Capabilities: [90] MSI: Enable- Count=1/4 Maskable- 64bit+
        Capabilities: [c4] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Kernel modules: xhci_pci

03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)
        Subsystem: Gigabyte Technology Co., Ltd Onboard Ethernet
        Flags: bus master, fast devsel, latency 0, IRQ 31, NUMA node 0, IOMMU group 16
[... cut ...]
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, MSI 01
        Capabilities: [b0] MSI-X: Enable+ Count=4 Masked-
        Capabilities: [d0] Vital Product Data
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [140] Virtual Channel
        Capabilities: [160] Device Serial Number 01-00-00-00-68-4c-e0-00
        Capabilities: [170] Latency Tolerance Reporting
        Kernel driver in use: r8169
        Kernel modules: r8169

Calimerorulez, 2024-05-05:
What is the output of 

`lsmod | grep apex`

jackj120, 2024-05-05:
Nothing

Calimerorulez, 2024-05-05:
@jackj120 

Then the driver you just compiled and installed isn't loaded.

Did you reboot?

jackj120, 2024-05-05:
@Calimerorulez rebooted for the second time and nothing.

Calimerorulez, 2024-05-05:
@jackj120 

Then I currently out of options too I'm afraid...

Maybe I'll think up something later

jackj120, 2024-05-05:
@Calimerorulez Thank you anyway... much appreciated

magic-blue-smoke, 2024-05-06:
@jackj120  adapter board has PCIe switch and Coral TPUs are behind it, ie we should see it first. With lspci output provided, I can't see ASMedia device. 
On a host machine, could you please show output of ```lspci -nn | grep 1182```?
If nothing, check if adapter is properly seated in PCIe slot and not reversed (sometimes this happens)
If it won't work, please contact me using [form at the bottom of the page](https://mbs-shop.online/) for adapter replacement

jackj120, 2024-05-07:
@magic-blue-smoke 
Just tried  ### lspci -nn | grep 1182  ### but nothing and yes the adapter is properly seated . I've tried multiple times reboot ,connect and disconnect .

rice1204, 2024-05-08:
same issue here on a PCIe card I received last week.
Will send an email.

jackj120, 2024-12-10:
Hi, after a few months, I decided to buy another adapter, and now it works great. I can see both Corals, and Frigate is running fine. Thanks @magic-blue-smoke!

closed 2024-12-10 by jackj120