magic-blue-smoke/Dual-Edge-TPU-Adapter #59: Minisforum MS-01 compatibility

Doug411, 2024-05-07:
I have a Minisform MS-01.  It has the following B&M key NVME slots

M.2 2280 NVMe SSD slot (Alt U.2) (PCIe 4.0 x4): One slot
M.2 2280/22110 NVMe SSD slot (PCIe 3.0 x4): One slot
M.2 2280/22110 NVMe SSD slot (PCIe 3.0 x2): One slot

Would I be able to use the dual edge TPU with your b&m adapter in the 3X2 slot and have both TPUs recognized?

It also has an e key for wifi/Bluetooth.  But I believe I read that it doesn't support bifurcation.

jianhuiz, 2024-05-07:
I'm using the 3.0x2 slot and have both TPUs recognized and they work perfectly.

However, the tricky part is that there is not enough space under the SSD fan to fit the adapter and the TPU card, I have to raise the fan and cut a small piece of the metal plate to fit everything back into the chassis.

Doug411, 2024-05-07:
What inference speed are you getting? 

UGH... How did you raise the fan?  I'm trying to visualize why you would need to cut the plate.  Is the plate too low as well?

Im visiting in a location where I don't have my tools.  What did you use to cut the plate?

Doug411, 2024-05-07:
oh, and how did you raise the fan?

Doug411, 2024-05-07:
Also what do you think of using an extender?  Do you think there is space to reroute a little?   I wonder if the 3X4 extender would work in the 3X2 slot?

:[
](https://www.amazon.com/gp/product/B07DZCCGJN/ref=ox_sc_act_title_3?smid=A3LUEZHQOA50VU&th=1)

or 
[
](https://www.amazon.com/gp/product/B08XXDPMWK/ref=ox_sc_act_title_2?smid=A69X5KMJHM850&psc=1)

jianhuiz, 2024-05-08:
I get 7.9~8.1ms inference speed.

I cut a pen refill tube into 2~3mm pieces as spacers.
I also needed to cut the top corner in this picture after raised the fan or else it's in the way when slide the cover back.
The metal is pretty soft, I used a pair of pliers, a flush cutter and a knife to get a better looking.
https://www.servethehome.com/minisforum-ms-01-review-the-10gbe-with-pcie-slot-mini-pc-intel/minisforum-ms-01-internal-overview-ssd-and-wifi-side-with-cooler/

I doubt there is enough space using an extender, basically you'll end up with the extender board in the slot, an cable and three stacked PCBs the other end.

Doug411, 2024-05-08:
I wish I would have posted my question a few weeks ago!  My TPU arrives today... But I have to wait a few weeks for my magic-blue-smoke adapter!  I'll be very happy with those speeds.  I was pessimistic about it recognizing both TPUs.  That's great!  

 I'm assuming you raised the whole metal shrowd(not just the fan?) .  Without taking my machine apart right now, I'm visualizing the tubes you cut resting on top of the screw posts on the board.   And the screws that attach the shrowd fit inside  the tubes/spacers, enabling them to screw into the posts with some small clearance.  Is that correct?   Can you circle the piece that you had to cut on the shrowd?   I'm trying to visualize why one corner had to be cut.  Sounds like your cut didn't interfere with the screw holes that attach the shrowd to the board?

Doug411, 2024-05-08:
If I described it correctly, and you put the spacers on top of the screw posts, did you just put them on top of the one corner, or all corners?   If you just raised the one corner, could you have bent it down versus cutting it?

jianhuiz, 2024-05-08:
I was reluctant climbing up to disconnect the box (another flaw that the ethernet cable is supper hard to pull out due to the tight space), but let me do it now and get you some pictures.

jianhuiz, 2024-05-08:
All corners. Bend the plate can block the fan as well.
It blocks the chassis from closing because it's higher now, almost touching the back cover.

[image]
[image]
[image]

Doug411, 2024-05-09:
Thanks so much! That helps.   Nicely done.    I did end up buying one of those extenders off of Amazon.  It's not very long.   I was hoping I could bend/twist the ribbon and have the coral and extender card sit where there is space for the U.2 drive.   However I'm no electronics wizard.  I'm afraid of it shorting the motherboard.  I'm not sure how to attach it so that it doesn't cause harm (not sure It's long enough anyway) .  I'll probably end of doing the exact same thing you did.

Doug411, 2024-06-29:
I ended up performing your mods on my machine.   I super glued the standoffs, and that was a pain.  You were correct, cutting through the shrowd was no problem at all.

Are you running Proxmox?    I haven't been able to get mine to work.   I initially tried my coral in the wifi card slot (knowing that I would only see 1 TPU).  However, that didn't work at all.  The TPU was not showing up at all, and the card caused some some contention issues.

I got my adapter and installed it in the m.2 slot, now I can see both TPUs in LSPCI (progress).  However, now, I lose my Ethernet connection on the i226LM port.  Did you have any issues, and if so, how did you solve.    I fear maybe I have a bad card.  I noticed some of my pci devices move around when I have the adapter in the m.2 slot.  Scroll to the end below.  This is what it looks like without the card.   When I put in the card, my TPUs show up as 5b:00.0 and 5c:00.0 and my recollection is that my ethernet management port, Ethernet controller: Intel Corporation Ethernet Controller I226-LM (rev 04),  moves from 59:00.0 to 5d:00.0.

00:00.0 Host bridge: Intel Corporation 12th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.1 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.2 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.3 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.4 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.5 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.6 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.7 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:06.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 (rev 02)
00:06.2 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #2 (rev 02)
00:07.0 PCI bridge: Intel Corporation Alder Lake-P Thunderbolt 4 PCI Express Root Port #0 (rev 02)
00:07.2 PCI bridge: Intel Corporation Alder Lake-P Thunderbolt 4 PCI Express Root Port #2 (rev 02)
00:0d.0 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 USB Controller (rev 02)
00:0d.2 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 NHI #0 (rev 02)
00:0d.3 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 NHI #1 (rev 02)
00:14.0 USB controller: Intel Corporation Alder Lake PCH USB 3.2 xHCI Host Controller (rev 01)
00:14.2 RAM memory: Intel Corporation Alder Lake PCH Shared SRAM (rev 01)
00:16.0 Communication controller: Intel Corporation Alder Lake PCH HECI Controller (rev 01)
00:16.3 Serial controller: Intel Corporation Alder Lake AMT SOL Redirection (rev 01)
00:1c.0 PCI bridge: Intel Corporation Alder Lake-P PCH PCIe Root Port (rev 01)
00:1c.4 PCI bridge: Intel Corporation Device 51bc (rev 01)
00:1d.0 PCI bridge: Intel Corporation Device 51b2 (rev 01)
00:1d.3 PCI bridge: Intel Corporation Device 51b3 (rev 01)
00:1f.0 ISA bridge: Intel Corporation Alder Lake PCH eSPI Controller (rev 01)
00:1f.3 Audio device: Intel Corporation Alder Lake PCH-P High Definition Audio Controller (rev 01)
00:1f.4 SMBus: Intel Corporation Alder Lake PCH-P SMBus Host Controller (rev 01)
00:1f.5 Serial bus controller: Intel Corporation Alder Lake-P PCH SPI Controller (rev 01)
01:00.0 Non-Volatile memory controller: Seagate Technology PLC FireCuda 520 SSD (rev 01)
02:00.0 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GbE SFP+ (rev 02)
02:00.1 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GbE SFP+ (rev 02)
57:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
58:00.0 Non-Volatile memory controller: Sandisk Corp WD Black SN770 NVMe SSD (rev 01)
59:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-LM (rev 04)
5a:00.0 Network controller: MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter

Doug411, 2024-06-29:
BTW, while I was waiting for the adapter I installed frigate using the openvino detector.  I also enabled hardware acceleration using SRIOV GPU pass through.   I'm running 5 wyze cameras doing detection at 1080p 5fps.   I'm getting the streams from the cameras through wyze-bridge (another layer of ffmpeg conversion to RTSP).   Frigate is running in a VM (not an LXC).   My inference times are  always under 9 (8.75 on average I would say).   I never expected it to be that good without the coral.   However, I would like to get mine to work regardless.   FYI, my ms-01 is the 12900h.   I first purchased a 13900h, and it was defective, and I was too impatient to wait for another 13900 so I took an available 12900.   Performance so far on everything has been superb.

Doug411, 2024-07-01:
I solved the network issue.   After installing the adapter, my i226LM port changed from enp89s0 to enp93s0.   I updated /etc/network/interfaces accordingly and  I am back in business.  Next step passing the TPUs through to frigate (Ubuntu VM).

jianhuiz, 2024-07-02:
I was using one of the 10G port, so didn't notice the device changes to i226LM.

I'm using PVE 8.1, passthrough the TPUs to Debian VM running frigate.
I didn't try SRIOV GPU detection but only configured it for decoding, qsv didn't work for me, vaapi works but was not stable: I got decode errors after a couple days. It's probably because of the Debian kernel is too old, going to try Ubuntu when I got time, or wait for 6.10/11 kernel comes out.

BTW, <9ms inferencing time using GPU is pretty good, if I knew it, probably wouldn't bother getting the TPUs.

My ms-01 is 12900H as well.