magic-blue-smoke/Dual-Edge-TPU-Adapter #62: Question:  is there any reason why the low profile PCIe x1 version wouldn't work with another device such as a Hailo-8L E-Key?

goatzillax, 2024-06-05:
I'm just curious if this same hardware can work with other m.2 accelerators.  Thanks!

magic-blue-smoke, 2024-06-06:
Hi @goatzillax 
TLDR:
It should work, but there's no reason doing so and you should expect performance drop

Some details:
To my best knowledge, Hailo-8L is a single chip solution and has one PCIe port with two Gen3 lanes, rather than two PCIe ports with one Gen2 lane each we have from Coral Dual Edge E-key card. This means no PCIe switch IC is needed and passive PCIe adapter from your local store should work without issues. I'd personally prefer B+M key Hailo-8L card along with m.2 M-key to PCIe x4 passive card for NVMe drives.
So, there's no need for PCIe switch, unless there's demand for dedicated multiple Hailo-8L card adapter board
Have to add that recommendations above are result of three minutes research without any guarantee of compatibility

goatzillax, 2024-06-06:
OK, thanks.  This was mostly just a question of just "rolling with the equipment you already (might) have".