magic-blue-smoke/Dual-Edge-TPU-Adapter #63: dual edge tpu + m.2 b+m

gustvao, 2024-08-02:
Hi-

I was wondering if it is possible to make these 2+1 tpus work in a pcie x8 with a splitter - wnated to confirm before buying:

stack would be:

- dual edge on this adapter: https://www.makerfabs.com/dual-edge-tpu-adapter-m2-2280-b-m-key.html
- m2 b+m w/o adapter
- both plugged on either:
https://www.aliexpress.com/item/1005006996504319.html
or
https://www.aliexpress.com/item/1005006930328261.html

or any other that might work from ali or amazon..

could you kindly point me to the right adaptors?

tks

rhclayto, 2024-11-07:
I have the same question.

apple-fritter, 2024-11-10:
Forgive this suggestion, but there is a single piece of kit, available soon from [here](https://mbs-shop.online/), from https://mbs-shop.online/.

[image]

apple-fritter, 2024-11-10:
Also, if you read the information about the B+M adapters you mention, they will not work in an M.2 slot that is SATA specific, and will not work in an external enclosure, irrespective of being NVME purposed.