magic-blue-smoke/Dual-Edge-TPU-Adapter #66: Coral M.2 Accelerator dual edge tpu with Dual Edge TPU Adapter - PCIe x1  -- Crashes PC

TheeDudenator, 2024-09-03:
I have this setup and it can work with Frigate.
But 1-2mins of detection will force a reboot on the machine.  On the screen there is a flash of "hardware error"
If I do not use Frigate with pcie, then the machine works fine.

If this would a heat issue, I would not think it would kill the machine this quick.

Linux Mint 22
Kernel 6.8.0-41

lspci -nn | grep 089a
06:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
07:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]

ls -l /dev/apex*
crw-rw---- 1 root apex 120, 0 Sep  3 13:14 /dev/apex_0
crw-rw---- 1 root apex 120, 1 Sep  3 13:14 /dev/apex_1

dmesg | grep apex
[    4.975759] apex 0000:06:00.0: enabling device (0000 -> 0002)
[    4.992796] apex 0000:07:00.0: enabling device (0000 -> 0002)
[   10.130259] apex 0000:06:00.0: Apex performance not throttled due to temperature
[   10.131238] apex 0000:07:00.0: Apex performance not throttled due to temperature

TheeDudenator, 2024-09-03:
Doing some testing.
It seems that if I disable the first TPU and just use the second, the machine does not crash.

If I use the first TPU and disable the second - it will crash.

magic-blue-smoke, 2024-09-03:
Hi @TheeDudenator 
Could you please try to:
- reinsert TPU card and clean contacts
- visually inspect TPU card to see if there are some cracks or scratches on TPUs
- let Frigate run for a couple of minutes, stop it and get some diagnostics with `lspci -vvv` before it crash

TheeDudenator, 2024-09-03:
Powered down the system
Removed PCI card and the TPU card
Cleaned all contacts with alcohol cloth
I could not visually see any issues.

Restarted with Frigate still only using the 1 TPU.
The one that does not crash.

00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Root Complex
	Subsystem: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Root Complex
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-

00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Starship/Matisse IOMMU
	Subsystem: Advanced Micro Devices, Inc. [AMD] Starship/Matisse IOMMU
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin ? routed to IRQ 26
	Capabilities: <access denied>

00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	IOMMU group: 0

00:01.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge (prog-if 00 [Normal decode])
	Subsystem: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin ? routed to IRQ 28
	IOMMU group: 1
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 0000f000-00000fff [disabled] [32-bit]
	Memory behind bridge: fcf00000-fcffffff [size=1M] [32-bit]
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled] [64-bit]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16+ MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: <access denied>
	Kernel driver in use: pcieport

00:01.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge (prog-if 00 [Normal decode])
	Subsystem: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin ? routed to IRQ 29
	IOMMU group: 2
	Bus: primary=00, secondary=02, subordinate=0d, sec-latency=0
	I/O behind bridge: d000-efff [size=8K] [16-bit]
	Memory behind bridge: fc400000-fcafffff [size=7M] [32-bit]
	Prefetchable memory behind bridge: e0300000-e06fffff [size=4M] [32-bit]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16+ MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: <access denied>
	Kernel driver in use: pcieport

00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	IOMMU group: 3

00:03.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	IOMMU group: 4

00:03.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge (prog-if 00 [Normal decode])
[... 499 lines cut ...]
	Kernel modules: ccp

10:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Matisse USB 3.0 Host Controller (prog-if 30 [XHCI])
	Subsystem: Gigabyte Technology Co., Ltd Matisse USB 3.0 Host Controller
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin C routed to IRQ 112
	IOMMU group: 33
	Region 0: Memory at fcb00000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: <access denied>
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci

10:00.4 Audio device: Advanced Micro Devices, Inc. [AMD] Starship/Matisse HD Audio Controller
	Subsystem: Gigabyte Technology Co., Ltd Starship/Matisse HD Audio Controller
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin D routed to IRQ 187
	IOMMU group: 34
	Region 0: Memory at fcd00000 (32-bit, non-prefetchable) [size=32K]
	Capabilities: <access denied>
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel

TheeDudenator, 2024-09-03:
I disabled TPU1 and enabled TPU2
I did not get a chance to copy/paste lspci -vvv
Its not that Frigate crashes, the entire machine boots back to BIOS screen.
It does flash some txt, that I can't see quick enough.
Maybe Linux Mint has a log I can look at.,

TheeDudenator, 2024-09-03:
After the crash I ran this.

journalctl -k -r -b -1 --lines=50

Sep 03 18:45:17 Basement kernel: r8169 0000:0a:00.0 enp10s0: NETDEV WATCHDOG: CPU: 14: transmit queue 0 timed out 9398 ms
Sep 03 18:45:06 Basement kernel: usb 5-6.2: USB disconnect, device number 3
Sep 03 18:45:06 Basement kernel: usb 5-6: USB disconnect, device number 2
Sep 03 18:45:06 Basement kernel: xhci_hcd 0000:0b:00.3: HC died; cleaning up
Sep 03 18:45:06 Basement kernel: xhci_hcd 0000:0b:00.3: xHCI host controller not responding, assume dead
Sep 03 18:44:58 Basement kernel: megaraid_sas 0000:08:00.0: Outstanding fastpath IOs: 0
Sep 03 18:44:58 Basement kernel: megaraid_sas 0000:08:00.0: resetting fusion adapter scsi0.
Sep 03 18:44:58 Basement kernel: megaraid_sas 0000:08:00.0: FW in FAULT state Fault code:0xfff0000 subcode:0xff00 func:megasas_wait_for_outstanding_fusion
Sep 03 18:44:58 Basement kernel: megaraid_sas 0000:08:00.0: megasas_disable_intr_fusion is called outbound_intr_mask:0xffffffff
Sep 03 18:44:56 Basement kernel: apex 0000:07:00.0: Apex performance not throttled due to temperature
Sep 03 18:43:01 Basement kernel: br-c71b9fa458a7: port 1(veth2d7e804) entered forwarding state
Sep 03 18:43:01 Basement kernel: br-c71b9fa458a7: port 1(veth2d7e804) entered blocking state
Sep 03 18:43:01 Basement kernel: eth0: renamed from veth2f54fc3
Sep 03 18:43:01 Basement kernel: veth2d7e804: entered promiscuous mode
Sep 03 18:43:01 Basement kernel: veth2d7e804: entered allmulticast mode
Sep 03 18:43:01 Basement kernel: br-c71b9fa458a7: port 1(veth2d7e804) entered disabled state
Sep 03 18:43:01 Basement kernel: br-c71b9fa458a7: port 1(veth2d7e804) entered blocking state
Sep 03 18:43:01 Basement kernel: br-c71b9fa458a7: port 1(vetha61ba86) entered disabled state
Sep 03 18:43:01 Basement kernel: vetha61ba86 (unregistering): left promiscuous mode
Sep 03 18:43:01 Basement kernel: vetha61ba86 (unregistering): left allmulticast mode
Sep 03 18:43:01 Basement kernel: br-c71b9fa458a7: port 1(vetha61ba86) entered disabled state
Sep 03 18:43:01 Basement kernel: br-c71b9fa458a7: port 1(vetha61ba86) entered disabled state
Sep 03 18:43:01 Basement kernel: vethf89f9fc: renamed from eth0
Sep 03 18:36:57 Basement systemd-journald[539]: Time jumped backwards, rotating.
Sep 03 18:36:47 Basement kernel: logitech-hidpp-device 0003:046D:4082.0005: HID++ 4.5 device connected.
Sep 03 18:36:44 Basement systemd-journald[539]: /var/log/journal/c3c996e5bc3a40599010b84aefd0aedc/user-1000.journal: Journal file uses a different sequence number ID, rotating.
Sep 03 18:36:43 Basement kernel: logitech-hidpp-device 0003:046D:408A.0006: HID++ 4.5 device connected.
Sep 03 18:36:38 Basement kernel: br-c71b9fa458a7: port 1(vetha61ba86) entered forwarding state
Sep 03 18:36:38 Basement kernel: br-c71b9fa458a7: port 1(vetha61ba86) entered blocking state
Sep 03 18:36:38 Basement kernel: eth0: renamed from vethf89f9fc
Sep 03 18:36:38 Basement kernel: docker0: port 1(vethff5a6cb) entered forwarding state
Sep 03 18:36:38 Basement kernel: docker0: port 1(vethff5a6cb) entered blocking state
Sep 03 18:36:38 Basement kernel: eth0: renamed from veth9175152
Sep 03 18:36:38 Basement kernel: vetha61ba86: entered promiscuous mode
Sep 03 18:36:38 Basement kernel: vetha61ba86: entered allmulticast mode
Sep 03 18:36:38 Basement kernel: br-c71b9fa458a7: port 1(vetha61ba86) entered disabled state
Sep 03 18:36:38 Basement kernel: br-c71b9fa458a7: port 1(vetha61ba86) entered blocking state
Sep 03 18:36:38 Basement kernel: vethff5a6cb: entered promiscuous mode
Sep 03 18:36:38 Basement kernel: vethff5a6cb: entered allmulticast mode
Sep 03 18:36:38 Basement kernel: docker0: port 1(vethff5a6cb) entered disabled state
Sep 03 18:36:38 Basement kernel: docker0: port 1(vethff5a6cb) entered blocking state
Sep 03 18:36:38 Basement kernel: Initializing XFRM netlink socket
Sep 03 18:36:37 Basement kernel: Bridge firewalling registered
Sep 03 18:36:37 Basement kernel: bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
Sep 03 18:36:37 Basement kernel: evm: overlay not supported
Sep 03 18:36:37 Basement kernel: audit: type=1400 audit(1725402997.872:120): apparmor="STATUS" operation="profile_load" profile="unconfined" name="docker-default" pid=1620 comm="apparmor_parser"
Sep 03 18:36:37 Basement kernel: kauditd_printk_skb: 108 callbacks suppressed
Sep 03 18:36:33 Basement kernel: r8169 0000:0a:00.0 enp10s0: Link is Up - 2.5Gbps/Full - flow control off
Sep 03 18:36:29 Basement kernel: apex 0000:07:00.0: Apex performance not throttled due to temperature
Sep 03 18:36:29 Basement kernel: apex 0000:06:00.0: Apex performance not throttled due to temperature
Sep 03 18:36:28 Basement kernel: r8169 0000:0a:00.0 enp10s0: Link is Down
Sep 03 18:36:28 Basement kernel: RTL8226B_RTL8221B 2.5Gbps PHY r8169-0-a00:00: attached PHY driver (mii_bus:phy_addr=r8169-0-a00:00, irq=MAC)

TheeDudenator, 2024-09-04:
After doing all that, now either TPU crashes.
I pulled your PCI adapter from my system.

magic-blue-smoke, 2024-09-11:
@TheeDudenator could you please contact me using form [at the bottom of the page](https://mbs-shop.online/) for further diagnostics and possible replacement/refund