magic-blue-smoke/Dual-Edge-TPU-Adapter #69: Boot issues with Low profile PCIe x1 version in HP DL360 Gen9

DavidValeri, 2024-11-30:
I received the Low profile PCIe x1 version this week. On installation in a HP DL360 Gen9, the system will not power on and reports the following issue:

Server Critical Fault (Service Information: Power On Fault, System Board,  P5V/P3V3/Chipset/AUX Regulators 1 (02h))

I tried the card with and without the TPU attached. I tried the card in two different PCIe slots. I know at least one of these slots is functional as it had a NIC in it until last week. Removing the card clears the error and the system boots as normal. System firmware is up to date based on my research.

Any ideas?

magic-blue-smoke, 2024-12-01:
Hi @DavidValeri 
Sorry to hear about the issue you have. Some servers are not compatible with ASMedia PCIe switch IC used in adapter. My best guess is there's something BIOS doesn't like at POST and PCIe init
Despite of this what I can propose is:
- verify that adapter is not reversed. Bracket mount holes should be facing towards motherboard/riser external edge. This is really rare and simple mistake, but mentioned power error might be indication of this
- try adapter in desktop PC if it does work. If it does - please try using adapter in another server, another brand or another gen of the server if possible
- if there's no other use for adapter than mentioned server, please contact Makerfabs for return

DavidValeri, 2024-12-01:
🤦‍♂️ Well this embarrassing episode will forever be in the archives when somebody Googles me. I indeed flipped the card when I put it in the first time. Installed it the other way. It boots and lspci detects the TPUs.

closed 2024-12-01 by DavidValeri

apple-fritter, 2024-12-12:
@DavidValeri good to know it survived at least.