magic-blue-smoke/Dual-Edge-TPU-Adapter #70: I need an Adapter for Coral TPU Dual

emilianogetino, 2024-11-30:
Hello, I need an Adapter for Coral TPU Dual.
It is for a BKOUEN model CK11 miniPC.
The slot where I want to put it is this:

[image]

Could you please tell me which one I should buy and the link to it?

Greetings, thank you

magic-blue-smoke, 2024-12-01:
Hi @emilianogetino 
I'd need a link to motherboard manual to give a precise answer.
- we need to make sure this slot has PCIe interface
- we need to make sure there's enough clearance above to fit Coral board
- we need to make sure it's 2280 slot

If conditions above are met, you can use this adapter: https://www.makerfabs.com/dual-edge-tpu-adapter-m2-2280-b-m-key.html

emilianogetino, 2024-12-02:
Thanks for your response.
Space if there is and the slot if it is 2280.
What I don't know is if the slot has a PCIe interface.
Would there be any way to know?
The miniPC is the CK11 model, which is sold by different brands, mine is BKOUEN, I have tried to find the manual without success.
I have also written to the seller to see if he will send me the manual.

Greetings, thank you.

emilianogetino, 2024-12-11:
Good morning, I have tried to find a manual for the minipc as well as contact the seller and neither of the two things have been possible.
So I have started the mini PC with a Windows and I have collected information about it. If you can tell me if with your adapter I could use two TPUs of my Coral Dual.
The miniPc is a BKOUEN CK11.
It does not provide information about the motherboard, only the bios is from Americam megatrends and the model ck12v101, but I did not find any information about it either.

[image]
[image]
[image]

As for PCIe, it seems to indicate that there are 6 slots
[image]
[image]

However, there are only 3 slots on the motherboard:
- 1 NVEMe/NGFF, where my NVEMe HDD is installed
- 1 where the wifi card is installed, and where the dual coral does work, but only one of the two TPUs 
- 1 NGFF where I want to install the dual coral, and have both TPUs work

[image]

If I look in another section, it does seem to show 3 PCIe, although now I see that if I had pressed the "+" perhaps more would appear

[image]

In another section it tells me this:

[image]
[image]
[image]
[image]
[image]
[image]

With this data, could you tell me if with your adapter in the NFGG slot the 2 TPUs of the Google Coral would work for me?
Thank you for your time, thank you.
All the best.

apple-fritter, 2024-12-12:
> Hi @emilianogetino
> I'd need a link to motherboard manual to give a precise answer.
> - we need to make sure this slot has PCIe interface
> - we need to make sure there's enough clearance above to fit [...]

The slot says NGFF only.

emilianogetino, 2024-12-12:
Thanks for your time.
All the best.

apple-fritter, 2024-12-12:
@emilianogetino is your SSD SATA or NVMe? If it's SATA you could move it over to the NGFF only slot, and utilize the NVME slot with the B+M adapters from this project.

emilianogetino, 2024-12-12:
> @emilianogetino¿Tu SSD es SATA o NVMe? Si es SATA, puedes moverlo a la ranura NGFF únicamente y utilizar la ranura NVME con los adaptadores B+M de este proyecto.

My HDD is NVMe, unfortunately I can't move it between slots.
For now I will leave the Dual Coran in the wifi card slot, which I do not use, although this way only one of the TPUs works for me.
All the best.