magic-blue-smoke/Dual-Edge-TPU-Adapter #71: Dual TPU, PCI Adapter on Proxmox - no Coral device in lspci

GergoIO, 2024-12-12:
Hello, I'm hoping you can help with my dual edge TPU and PCI adapter.

I've tried all the ways I can think but I still can't get the device to show under lspci.
The contacts look clean and there's nothing obviously wrong with either the tpu or the adapter (from what I can tell).

[image]

[image]

I have tried to add the adapter to different PCI slots (link to [motherboard](https://www.msi.com/Motherboard/PRO-B760-P-DDR4-II/Specification)) and I get a different entry with lspci each time but never for the Coral device.
Interestingly, the new entry shows up also when I only add the adapter without the TPU installed.

With the adapter in **PCI E1** (00:01.0 is the new line that shows):
```
00:00.0 Host bridge: Intel Corporation 12th Gen Core Processor Host Bridge (rev 05)
***00:01.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x16 Controller #1 (rev 05)
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 730] (rev 0c)
00:06.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 (rev 05)
00:08.0 System peripheral: Intel Corporation 12th Gen Core Processor Gaussian & Neural Accelerator (rev 05)
00:0a.0 Signal processing controller: Intel Corporation Platform Monitoring Technology (rev 01)
00:14.0 USB controller: Intel Corporation Raptor Lake USB 3.2 Gen 2x2 (20 Gb/s) XHCI Host Controller (rev 11)
00:14.2 RAM memory: Intel Corporation Raptor Lake-S PCH Shared SRAM (rev 11)
00:16.0 Communication controller: Intel Corporation Raptor Lake CSME HECI (rev 11)
00:17.0 SATA controller: Intel Corporation Raptor Lake SATA AHCI Controller (rev 11)
00:1c.0 PCI bridge: Intel Corporation Raptor Lake PCI Express Root Port (rev 11)
00:1f.0 ISA bridge: Intel Corporation Device 7a06 (rev 11)
00:1f.3 Audio device: Intel Corporation Raptor Lake High Definition Audio Controller (rev 11)
00:1f.4 SMBus: Intel Corporation Raptor Lake-S PCH SMBus Controller (rev 11)
00:1f.5 Serial bus controller: Intel Corporation Raptor Lake SPI (flash) Controller (rev 11)
02:00.0 Non-Volatile memory controller: Kingston Technology Company, Inc. Device 5026
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
```

With the adapter in **PCI E3** (no more 00:01.0 but a new entry of 00:1c.4): 
```
00:00.0 Host bridge: Intel Corporation 12th Gen Core Processor Host Bridge (rev 05)
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 730] (rev 0c)
00:06.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 (rev 05)
00:08.0 System peripheral: Intel Corporation 12th Gen Core Processor Gaussian & Neural Accelerator (rev 05)
00:0a.0 Signal processing controller: Intel Corporation Platform Monitoring Technology (rev 01)
00:14.0 USB controller: Intel Corporation Raptor Lake USB 3.2 Gen 2x2 (20 Gb/s) XHCI Host Controller (rev 11)
00:14.2 RAM memory: Intel Corporation Raptor Lake-S PCH Shared SRAM (rev 11)
00:16.0 Communication controller: Intel Corporation Raptor Lake CSME HECI (rev 11)
00:17.0 SATA controller: Intel Corporation Raptor Lake SATA AHCI Controller (rev 11)
00:1c.0 PCI bridge: Intel Corporation Raptor Lake PCI Express Root Port (rev 11)
***00:1c.4 PCI bridge: Intel Corporation Device 7a3c (rev 11)
00:1f.0 ISA bridge: Intel Corporation Device 7a06 (rev 11)
00:1f.3 Audio device: Intel Corporation Raptor Lake High Definition Audio Controller (rev 11)
00:1f.4 SMBus: Intel Corporation Raptor Lake-S PCH SMBus Controller (rev 11)
00:1f.5 Serial bus controller: Intel Corporation Raptor Lake SPI (flash) Controller (rev 11)
01:00.0 Non-Volatile memory controller: Kingston Technology Company, Inc. Device 5026
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
```

lspci -nn | grep 089a
this returns nothing and I also get the following:
ls /dev/apex*
ls: cannot access '/dev/apex*': No such file or directory

I've also been through the process suggested [here](https://forum.proxmox.com/threads/update-error-with-coral-tpu-drivers.136888/#post-608975).

Any ideas or suggestions?

magic-blue-smoke, 2025-01-05:
Hi @GergoIO 
Sorry for the late reply 
With `lspci` you should at least see ASMedia ASM1182e PCIe switch IC. 
If it's not there, please check:
- if adapter is inserted correctly: TPU card mounting screw is looking towards motherboard edge, **not** towards inside
- `lspci` was tried on host OS
If all above looks good, please contact Makerfabs for board replacement

rice1204, 2025-01-30:
@GergoIO @continuousmining 
based on my own experience, my guess is this problem stems from your proxmox environment. I had to fight with different kernel versions and drivers a long time until I found a working combination. Prior to getting the drivers working on the proxmox host, I couldn't see the adapter in `lspci` either.

I pinned kernel 6.5.13-6 to get it running, but others have reported having some luck with 6.8.12-2 in the same [proxmox forum thread ](https://forum.proxmox.com/threads/update-error-with-coral-tpu-drivers.136888/) you linked. Did you read the thread all the way to the end? The instructions in your link won't work for newer kernels.

There have been a few forks of the kernel modules, depending on your kernel version.

continuousmining, 2025-01-31:
Yea that doesn’t help either. Lspci doesn’t show anything.  However the m.2 adapter in a pci adapter shows 1 tpu.    I think my card is dead