magic-blue-smoke/Dual-Edge-TPU-Adapter #72: M2 Dual Edge TPU + PCI Adapter on PowerEdge R720 - only one TPU detected

evandroid, 2024-12-21:
Hi, 

This issue is constantly being reported here, but all the solutions differs, so I choosed to open a new one.

I run a homelab system in a Dell R720 with 192GB ram and dual xeon CPU. Home assistant and frigate add-on in a VM (not LXC).

The host system is proxmox latest version (8.3.2)
Kernel version: 6.8.12-5-pve

As my intention always was passing through coral to HA VM, I did not compiled nothing or configured apex or gasket.

After follow instruções on coral website, all went "half" fine - I was able to detect one tpu and pass it into HA VM, where frigate also detected and it's using it. No problem here.

The real problem is, as R720 has no m2 or nvme slot, and PCI has not any kind of bifurcation, I bought the PCI adapter from magic-blue-smoke to solve this. But even with the switch chip also detected (ASMedia) , I can't get the two TPU. 

**Result of "lspci -tv" (relevant part)**

---------

+-[0000:40]-+-01.0-[41]--
 |           +-02.0-[42-45]----00.0-[43-45]--+-03.0-[44]----00.0  Global Unichip Corp. Coral Edge TPU
 |           |                               \-07.0-[45]--
 |           +-03.0-[46]--

--------------

It shows a "45" space where the other Global Unichip Corp. Coral Edge TPU should be appearing, but it's blank

**Output of "lspci --nn | grep Coral"**

44:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]

**Output of "lspci --nn | grep ASMedia"**

42:00.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch [1b21:1182]
43:03.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch [1b21:1182]
43:07.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch [1b21:1182]

**Output os "lspci --vvvv | grep Coral"** 

44:00.0 System peripheral: Global Unichip Corp. Coral Edge TPU (prog-if ff)
        Subsystem: Global Unichip Corp. Coral Edge TPU

**Content of my GRUB:**

GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=on iommu=pt pcie_acs_override=downstream,multifunction"

The ASMedia Chip and Coral TPU (the one detected) are on separated IOMMU groups.

Any help?

magic-blue-smoke, 2025-01-05:
Hi @evandroid 
Sorry for the late reply
Seeing ASMedia chip is a good sign, but one TPU is not what's expected here.
What I can recommend is:
- make sure to try lspci on host OS before using VMs 
- visually inspect TPU card if it has any signs of mechanical damage. TPUs and PMICs are bare flip-chips and easy to damage
- if possible, try another Coral card
- if all above is ok, please contact Makerfabs for board replacement

evandroid, 2025-01-08:
Hi! Bingo.

I just received a second coral dual edge tpu last week. I changed it and now it's working flawlessly with both tpus detected on proxmox host, passed through to guest (HAOS) directly and now Frigate are using both.

Too bad I lost my dispute timing on Ali express.

For all people struggling to use coral edge dual tpu on proxmox vms, just send a message, it's ridiculous easy to setup it up and pass it through directly to vm or lxc.

Thank you.

closed 2025-01-08 by evandroid