magic-blue-smoke/Dual-Edge-TPU-Adapter #73: No TPUs detected using adapter in Aoostar R1 (Intel N100)

Cold-Lemonade, 2025-01-04:
Using an adapter I installed an ssd in my m.2 wifi slot and I replaced my NMVe ssd in my other M.2 slot with the M.2 2280 Dual Edge TPU Adapter (with a Coral Dual Edge TPU seated in it). When I boot into Ubuntu 24.04 server, I don't see *any* tpus when I run ```lspci```.

The computer is an Aoostar R1 and according to the manufacturer's website the M.2 2280 slot is Gen 3 PCIe 3 x 2. I believe this hardware satisfies the requirements to use the Coral Dual Edge TPU.

I have checked the BIOS settings and there is nothing specific to the M.2 slot or PCIe more generally. I have disabled secure boot and I have read that that may cause trouble. But still I cannot detect any devices attached to the mini pc.

Any suggestions on what to try next?

Cold-Lemonade, 2025-01-04:
Some more information... To investigate whether my Coral Dual Edge TPU is the source of my problems, I inserted it into the wifi card slot and put an NVMe ssd in the m.2 2280 slot. If the Coral Dual Edge TPU is functioning, I believe one tpu should appear in the output of ```lspci``` and that is exactly what I find. So I believe the issue is either: (1) the m.2 2280 slot cannot support the Blue Smoke adapter; or (2) a defective Blue Smoke adapter. My m.2 2280 slot works fine with my BiWIN P3822 ssd 512GB--I think this confirms that the m.2 slot is functioning with at least PCIe Gen 3x2. So shouldn't it work the Coral Dual Edge TPU via the Blue Smoke adapter? Does this mean my Blue Smoke adapter is defective?

Cold-Lemonade, 2025-01-05:
More digging... With the BiWIN P3822 ssd in the m.2 2280 slot, I ran ```sudo lspci -vvv | grep -A 20 -i nvme```. The output shows:

```
01:00.0 Non-Volatile memory controller: MAXIO Technology (Hangzhou) Ltd. NVMe SSD Controller MAP1202 (DRAM-less) (rev 01) (prog-if 02 [NVM Express])
        Subsystem: MAXIO Technology (Hangzhou) Ltd. NVMe SSD Controller MAP1202
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 18
        IOMMU group: 12
        Region 0: Memory at 80a00000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable- Count=1/32 Maskable+ 64bit+
                Address: 0000000000000000  Data: 0000
                Masking: 00000000  Pending: 00000000
        Capabilities: [70] Express (v2) Endpoint, MSI 1f
                DevCap: MaxPayload 512 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 25W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+ FLReset-
                        MaxPayload 256 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 8GT/s, Width x4, ASPM L1, Exit Latency L1 <64us
--
        Kernel driver in use: nvme
        Kernel modules: nvme
```

On the "LnkCap:" line, I believe "Speed 8GT/s" indicates PCIe Gen 3 and "Width x4" means x4. So my m.2 2280 slot has one PCIe Gen 3 x 4. @magic-blue-smoke Isn't this what the Magic Blue Smoke m.2 adapter is designed for?

magic-blue-smoke, 2025-01-05:
Hi @Cold-Lemonade 
Trying NVMe SSD in m.2 slot confirms it has PCIe, rather than SATA. This means you should see at least PCIe switch IC, which is ASMedia ASM1182e, even with adapter only/without Coral card.
While all the boards are tested after manufacturing, we can't exclude DoA boards. If you can't see ASMedia chip in `lspci -tvvv`, please contact Makerfabs for adapter replacement

magic-blue-smoke, 2025-01-19:
@Cold-Lemonade there definitely should be 1182 device under lspci on a host system. Please let me know the results with another PC