magic-blue-smoke/Dual-Edge-TPU-Adapter #77: PCIe x1 Gen2 Packet Switch Not Detected After Crash in ROMED8-2T Mobo

nonamenonumber, 2025-04-13:
Hi,
Bit of a stange issue here.
I've recently migrated from a consumer machine to a proper server mobo setup (ROMED8-2T, EPYC 7313).

Adapter and dual TPU working fine previously.

First boot in new machine looked fine, could see the ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch and 2x Coral TPUs in lspci.

When i first booted a VM with one of the Corals passed through, the system crashed and proceeded to get stuck in a reboot loop.
I could see several of the following errors in the IPMI:

 ID  |       TimeStamp      |    Sensor Name   |             Sensor Type            |                          Description                           
======|======================|==================|====================================|================================================================
 36   | 04/13/2025, 03:08:16 | BIOS             | Critical Interrupt                 | PCI SERR - Asserted
      |                      |                  |                                    | PCI Bus number : 0
      |                      |                  |                                    | PCI Device number : 3, PCI Function number : 1
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 35   | 04/13/2025, 03:06:47 | BIOS             | Critical Interrupt                 | PCI SERR - Asserted
      |                      |                  |                                    | PCI Bus number : 0
      |                      |                  |                                    | PCI Device number : 3, PCI Function number : 1
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 34   | 04/13/2025, 03:03:42 | BIOS             | Critical Interrupt                 | PCI SERR - Asserted
      |                      |                  |                                    | PCI Bus number : 0
      |                      |                  |                                    | PCI Device number : 3, PCI Function number : 1
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 33   | 04/13/2025, 03:02:15 | BIOS             | Critical Interrupt                 | PCI SERR - Asserted
      |                      |                  |                                    | PCI Bus number : 0
      |                      |                  |                                    | PCI Device number : 3, PCI Function number : 1
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 32   | 04/13/2025, 03:00:46 | BIOS             | Critical Interrupt                 | PCI SERR - Asserted
      |                      |                  |                                    | PCI Bus number : 0
      |                      |                  |                                    | PCI Device number : 3, PCI Function number : 1
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 31   | 04/13/2025, 02:59:56 | BIOS             | Critical Interrupt                 | PCI PERR - Asserted
      |                      |                  |                                    | PCI Bus number : 6
      |                      |                  |                                    | PCI Device number : 3, PCI Function number : 0
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 30   | 04/13/2025, 02:39:21 | BIOS             | Critical Interrupt                 | PCI PERR - Asserted
      |                      |                  |                                    | PCI Bus number : 5
      |                      |                  |                                    | PCI Device number : 0, PCI Function number : 0
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 29   | 04/13/2025, 02:39:21 | BIOS             | Critical Interrupt                 | PCI PERR - Asserted
      |                      |                  |                                    | PCI Bus number : 5
      |                      |                  |                                    | PCI Device number : 0, PCI Function number : 0
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 28   | 04/13/2025, 02:32:18 | BIOS             | Critical Interrupt                 | PCI PERR - Asserted
      |                      |                  |                                    | PCI Bus number : 5
      |                      |                  |                                    | PCI Device number : 0, PCI Function number : 0
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 27   | 04/13/2025, 02:32:17 | BIOS             | Critical Interrupt                 | PCI PERR - Asserted
      |                      |                  |                                    | PCI Bus number : 5
      |                      |                  |                                    | PCI Device number : 0, PCI Function number : 0
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 26   | 04/13/2025, 02:31:26 | BIOS             | Critical Interrupt                 | PCI SERR - Asserted
      |                      |                  |                                    | PCI Bus number : 0
      |                      |                  |                                    | PCI Device number : 3, PCI Function number : 1
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 25   | 04/13/2025, 02:29:58 | BIOS             | Critical Interrupt                 | PCI SERR - Asserted
      |                      |                  |                                    | PCI Bus number : 0
      |                      |                  |                                    | PCI Device number : 3, PCI Function number : 1
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 24   | 04/13/2025, 02:28:29 | BIOS             | Critical Interrupt                 | PCI SERR - Asserted
      |                      |                  |                                    | PCI Bus number : 0
      |                      |                  |                                    | PCI Device number : 3, PCI Function number : 1
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 23   | 04/13/2025, 02:27:00 | BIOS             | Critical Interrupt                 | PCI SERR - Asserted
      |                      |                  |                                    | PCI Bus number : 0
      |                      |                  |                                    | PCI Device number : 3, PCI Function number : 1
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 22   | 04/13/2025, 02:25:30 | BIOS             | Critical Interrupt                 | PCI SERR - Asserted
[... 16 lines cut ...]
      |                      |                  |                                    | PCI Bus number : 5
      |                      |                  |                                    | PCI Device number : 0, PCI Function number : 0
------|----------------------|------------------|------------------------------------|----------------------------------------------------------------
 17   | 04/13/2025, 02:20:05 | BIOS             | Critical Interrupt                 | PCI PERR - Asserted
[... cut ...]
I have tried moving the card to different slots (all 16x) and even specifying gen 2 pcie on the slot used, but it is no longer be detected at all.

I have reverted to a previous jank setup of putting the coral in an m2 adapter, that is then inserted in a pcie adapter. This works, and recognises a single Coral TPU fine.

I have put the dual edge adapter in the old machine, and that can successfully detect the ASM1182e in lspci. So it appears to me that both components are still working.

Bit of a long shot, but any ideas about what the incompatability might be?

Many thanks for the great product, has been rock solid for over a year

ptr727, 2025-06-23:
I also get [critical interrupt](https://github.com/magic-blue-smoke/Dual-Edge-TPU-Adapter/issues/78) errors reported on an AMD Epyc Supermicro server mobo.

I did try an A/E to M.2 adapter + M.2 to PCIe adapter, but the TPU was not detected, with this adapter it is detected, but the board reports critical interrupt errors.

Can you link to the other A/E adapter you used?

ptr727, 2025-06-23:
Weird, I tried something similar; https://www.amazon.com/dp/B07WBXP632 and https://www.amazon.com/dp/B09NKTYFHX and not detected, but with the makerfab adapter it was detected.

I had a Hailo8 in one slot and the Coral in the other, Hailo detected Coral not. I now use two separate cards for Hailo and Coral.