magic-blue-smoke/Dual-Edge-TPU-Adapter #80: Coral M.2 Dual Edge TPU with Dual Edge TPU Adapter crashes Unraid machine shortly after boot

nmur, 2025-07-30:
Tried installing a Coral M.2 with the Dual Edge TPU PCIe x1 Adapter, and have been finding that it crashes my entire system in under a minute or so after installing the [Coral plugin](https://github.com/unraid/unraid-coral-driver).

Looking at the plugin page shows that it's overheating pretty fiercely, despite an ambient system temperature of 20-30C and nothing being configured to utilise it yet. Immediately after the machine is shutdown, the Coral TPU feels barely warm. I am not using a heatsink of any kind yet

Plugin page immediately after installing plugin:

[image]

Plugin page after around 30 seconds and immediately before system crashes:

[image]

Logs from Unraid system:
```
Jul 30 00:51:58 Tower plugin-manager: running: 'anonymous'
Jul 30 00:51:58 Tower plugin-manager: creating: /boot/config/plugins/coral-driver/coral-driver-2025.03.25.txz - downloading from URL https://github.com/unraid/unraid-coral-driver/raw/master/packages/coral-driver-2025.03.25.txz
Jul 30 00:51:58 Tower plugin-manager: checking: /boot/config/plugins/coral-driver/coral-driver-2025.03.25.txz - MD5
Jul 30 00:51:58 Tower plugin-manager: running: upgradepkg --install-new /boot/config/plugins/coral-driver/coral-driver-2025.03.25.txz
Jul 30 00:51:59 Tower plugin-manager: creating: /usr/local/emhttp/plugins/coral-driver/README.md - from INLINE content
Jul 30 00:51:59 Tower plugin-manager: running: 'anonymous'
Jul 30 00:52:02 Tower kernel: apex 0000:08:00.0: enabling device (0000 -> 0002)
Jul 30 00:52:03 Tower kernel: apex 0000:09:00.0: enabling device (0000 -> 0002)
Jul 30 00:52:04 Tower plugin-manager: coral-driver.plg installed
Jul 30 00:52:09 Tower kernel: apex 0000:09:00.0: Apex performance not throttled due to temperature
Jul 30 00:52:09 Tower kernel: apex 0000:08:00.0: Apex performance not throttled due to temperature
Jul 30 00:52:14 Tower kernel: apex 0000:08:00.0: Apex performance not throttled due to temperature
Jul 30 00:52:19 Tower kernel: apex 0000:08:00.0: Apex performance not throttled due to temperature
Jul 30 00:52:24 Tower kernel: apex 0000:08:00.0: Apex performance not throttled due to temperature
Jul 30 00:52:29 Tower kernel: apex 0000:08:00.0: Apex performance not throttled due to temperature
Jul 30 00:52:34 Tower kernel: apex 0000:08:00.0: Apex performance not throttled due to temperature
Jul 30 00:52:40 Tower kernel: apex 0000:08:00.0: Apex performance not throttled due to temperature
Jul 30 00:52:45 Tower kernel: apex 0000:08:00.0: Apex performance not throttled due to temperature
Jul 30 00:52:45 Tower kernel: apex 0000:09:00.0: Apex performance not throttled due to temperature
```

I've re-inserted the M.2 TPU to the adapter a few times, and tried different PCIe x1 sockets. No obvious physical damage on the unit or adapter. 

Uninstalling the Coral plugin from Unraid allows the system to boot fine, as does just removing the adapter.

pocki, 2025-07-30:
My Setup is not so much newer: 
CPU: i5-11600
Motherboard: ASUS PRIME H570M-PLUS

I have some cheap heatsinks on it, about 40x40mm with some pads between the chips and heatsink. No active cooling for TPUs. 
Everthing is in a 4U Case together with 6 HDDs, 2 M2,... which is within a 19" Rack. 

I use the Corals in at Frigate (Docker Container) with 5 cameras and this settings: 
```
detectors:
  coral1:
    type: edgetpu
    device: pci:0
  coral2:
    type: edgetpu
    device: pci:1

model:
  path: /config/efficientdet_lite3_512_ptq_edgetpu.tflite
```

I'm sorry. I could not find my setup notes, so maybe the following may not be complete. 
I have the limetech version of this plugin installed: https://unraid.net/community/apps?q=Coral+Accelerator+Module+Drivers+#r
I found most the the configuration at frigate https://forums.unraid.net/topic/98064-support-blakeblackshear-frigate/ but it looks like the change something at the docs.
If you also want to use the TPUs in Docker Containers, you also need to map the device to the container similar to this: 
``` --device /dev/apex_0:/dev/apex_0 --device /dev/apex_1:/dev/apex_1 ```

nmur, 2025-07-31:
Thanks for the notes, should come in handy eventually.

I used the same plugin, but as I mentioned, the temperatures of both TPUs apparently reach shutdown levels (>104.8C) in under a minute whilst being completely idle and without Frigate or anything else configured to actually use them. 

So I'm wondering if this is usual behaviour for these module and a heatsink is absolutely necessary, or if there's a PCIe power delivery issue occurring, or if there's some faulty hardware involved here.

nmur, 2025-08-11:
I ended up getting one of [these M.2 2230 heatsinks](https://www.aliexpress.com/item/1005008053343723.html) and the difference seems to be night and day. Both TPUs are now in use in Frigate (just one camera so far) and temperatures are look excellent.

[image]